jedeschaud/ct_icp #6: how can i use kitti odometry data？

dongfangzhou1108, 2021-12-22:
hi，appreciate your great work，
but when i use kitti data，which have no distortion in http://www.cvlibs.net/datasets/kitti/eval_odometry.php，
that is mean there is no use to Update keypoints when run CT_ICP_GN（） function，
and i aslo think we do not need to use pose and velocity constraint in the paper，
what should i do to modify the code？

thanks a lot

GraceMelon, 2021-12-22:
I have a similar question. Since you mentioned the following for Kitti-corrected dataset:
"The frames are motion-compensated (no relative-timestamps) and the Continuous-Time aspect of CT-ICP will not work on this dataset." 
How to set the config file or modify the code to achieve the results you reported in your paper for Kitti-corrected dataset? I have a kitti-corrected dataset at hand. I tried to set the dataset_options.dataset to Kitti, the result is worse than the one reported in the paper. But when I set to Kitti-raw, the result is close to the one reported in the paper. That's quite confusing. 

Another question, one of the dataset I downloaded from Kitti website the synced lidar data is stored in .bin format while the raw lidar data is stored in .txt format. How to transform the .txt format to .ply?
Thanks!

GraceMelon, 2021-12-22:
> 
Please correct me if I'm wrong.
Update keypoints part in CT_ICP_GN() function is transforming the points to the map coordinate and is necessary for the next iteration.
I also doubt the improvement of adding pose and velocity constraint. I tried to comment Line 891 to Line 909 in ct_icp.cpp, the result didn't make a big difference.

pdell-kitware, 2021-12-22:
> I have a similar question. Since you mentioned the following for Kitti-corrected dataset: "The frames are motion-compensated (no relative-timestamps) and the Continuous-Time aspect of CT-ICP will [...]

Hi, 

To run CT-ICP on the KITTI corrected dataset, 
Which are motion corrected, you should specify the following options, in the YAML config file

odometry_options:
     motion_compensation: NONE

This will result in using the standard point to plane error (which optimizes on one pose as parameter), 
Instead of the two in our Continuous ICP. 

I will check, but normally you should have the same / close results as in our paper.

For the PLY files you can directly download the KITTI ply files we propose in the download link.

GraceMelon, 2021-12-22:
> hi，appreciate your great work， but when i use kitti data，which have no distortion in [...]

Please correct me if I'm wrong.
Update keypoints part in CT_ICP_GN() function is transforming the points to the map coordinate and is necessary for the next iteration.
I also doubt the improvement of adding pose and velocity constraint. I tried to comment Line 891 to Line 909 in ct_icp.cpp, the result doesn't make a big difference.

pdell-kitware, 2021-12-22:
> Please correct me if I'm wrong. Update keypoints part in CT_ICP_GN() function is transforming the points to the map coordinate and is necessary for the next iteration. I also doubt the improvement [...]

Well, we found that most of the time, these constraints are not needed (when there is enough geometric constraints in the environment), 
But sometimes, introducing two poses as parameters introduces too many degrees of freedom, which lead to failed / inconsistent poses. 
Think of it as an elastic cord which you hold by the two ends. If your constraints are only at the center of the cord, then there is a lot of positions in which you can place your two hands. If however your constraints are well spread between the beginning, and (where your hands are) and the middle, then you have a much better defined objective.

In our case the constraints are defined by the geometric features of the environment, so when you are in the first case (geometric constraints in the temporal middle of the frame), then these constraints are helpful to prevent misalignments.  
 
Hope this answer your question. If not, I will try to answer the best I can.

We are working hard to improve CT-ICP (notably introducing the ROS wrapping, which should be pushed in later this week)
So thanks for your comments, this will help us improve the interface further.

dongfangzhou1108, 2021-12-23:
> 

thanks for all your relpy，for run kitti correct，i try to modify the ct_icp_gn() function,
i delet all alpha_timestamp in 

                    // double cbx =
                    //         (1 - alpha_timestamp) * (frame_idx_previous_origin_begin[1] * closest_pt_normal[2] -
                    //                                  frame_idx_previous_origin_begin[2] * closest_pt_normal[1]);
                    // double cby =
                    //         (1 - alpha_timestamp) * (frame_idx_previous_origin_begin[2] * closest_pt_normal[0] -
                    //                                  frame_idx_previous_origin_begin[0] * closest_pt_normal[2]);
                    // double cbz =
                    //         (1 - alpha_timestamp) * (frame_idx_previous_origin_begin[0] * closest_pt_normal[1] -
                    //                                  frame_idx_previous_origin_begin[1] * closest_pt_normal[0]);

                    // double nbx = (1 - alpha_timestamp) * closest_pt_normal[0];
                    // double nby = (1 - alpha_timestamp) * closest_pt_normal[1];
                    // double nbz = (1 - alpha_timestamp) * closest_pt_normal[2];

                    // double cex = (alpha_timestamp) * (frame_idx_previous_origin_end[1] * closest_pt_normal[2] -
                    //                                   frame_idx_previous_origin_end[2] * closest_pt_normal[1]);
                    // double cey = (alpha_timestamp) * (frame_idx_previous_origin_end[2] * closest_pt_normal[0] -
                    //                                   frame_idx_previous_origin_end[0] * closest_pt_normal[2]);
                    // double cez = (alpha_timestamp) * (frame_idx_previous_origin_end[0] * closest_pt_normal[1] -
                    //                                   frame_idx_previous_origin_end[1] * closest_pt_normal[0]);

                    // double nex = (alpha_timestamp) * closest_pt_normal[0];
                    // double ney = (alpha_timestamp) * closest_pt_normal[1];
                    // double nez = (alpha_timestamp) * closest_pt_normal[2];

and delet the elastic constraint，
and make begin r/t same to end r/t，

dongfangzhou1108, 2021-12-23:
> Well, we found that most of the time, these constraints are not needed (when there is enough geometric constraints in the environment), But sometimes, introducing two poses as parameters introduces [...]

thanks a lot,
can i understand that the elastic constraint is for more accurate lidar points distortion？
i think your great work can perform de-distortion when optimite pose，
after add the well de-distorted cloud to the map, it well perform good because you build more accurate point clound map.

GraceMelon, 2021-12-28:
> Well, we found that most of the time, these constraints are not needed (when there is enough geometric constraints in the environment), But sometimes, introducing two poses as parameters introduces [...]

Thank you very much for your detailed explanation. Looking forward to the ROS wrapping.

closed 2022-01-13 by pdell-kitware