jedeschaud/ct_icp #7: libgflags link error

HeXu1, 2022-01-13:
Hi,thanks for your work
when I run "./slam -c robust_high_frequency_config.yaml" ,the output is "error while loading shared libraries: libgflags.so.2.2: cannot open shared object file: No such file or directory". I have install gflag in “usr/local/”. And I can't find where link gflags in cmakelists.txt.
hope for your reply,thanks a lot.

HeXu1, 2022-01-13:
it seems well now after install libgflags. but it has “Segmentation fault” when parse param with "OPTION_CLAUSE" function.

pdell-kitware, 2022-01-13:
Hi, 

Do you have an error associated with the Segmentation Fault (perhaps before) ?
Have you tried to debug (with a breakpoint) to know precisely where is the errror ?

I will need a bit more to be able to help you, 

Best Regards,

HeXu1, 2022-01-13:
thanks, 
Segmentation Fault  does not occur before.
Segmentation Fault comes from parse param part with "OPTION_CLAUSE" function. I have debug it,it cannot parse any param successfully.

pdell-kitware, 2022-01-13:
OPTION_CLAUSE is a MACRO, so you can look at the code it generates to see if there is any problem.

So i understand that it fails at  "OPTION_CLAUSE(icp_node, icp_options, threshold_voxel_occupancy, int);"
In the main script, is that right ?

Can you:
  - Give your system specifications (OS, compiler with version, etc..)
  - Print the YAML Node before calling the OPTION_CLAUSE

HeXu1, 2022-01-13:
ok,
my OS: Ubuntu16.04 
compiler with version :gcc8
And i print yaml node in "SLAMOptions read_config(const std::string &config_path)" function,the output is:
 slam_node: max_num_threads: 1
save_trajectory: true
suspend_on_failure: false
output_dir: .outputs
all_sequences: true
sequence: ""
start_index: 0
max_frames: -1
display_debug: true
dataset_options:
  dataset: NCLT
  root_path: /home/cidi/slam_data/ct-slam/2012-01-08
  fail_if_incomplete: false
  min_dist_lidar_center: 0.5
  max_dist_lidar_center: 100.0
  nclt_num_aggregated_pc: 220
odometry_options:
  voxel_size: 0.3
  sample_voxel_size: 1.5
  max_distance: 200.0
  max_num_points_in_voxel: 20
  debug_print: true
  min_distance_points: 0.15
  distance_error_threshold: 100.0
  motion_compensation: CONTINUOUS
  initialization: INIT_NONE
  init_num_frames: 20
  robust_registration: true
  robust_full_voxel_threshold: 0.6
  robust_num_attempts: 3
  robust_max_voxel_neighborhood: 4
  robust_threshold_relative_orientation: 2
  robust_threshold_ego_orientation: 2
  log_to_file: false
  debug_viz: true
  ct_icp_options:
    size_voxel_map: 0.8
    num_iters_icp: 30
    min_number_neighbors: 20
    threshold_voxel_occupancy: 5
    voxel_neighborhood: 1
    max_number_neighbors: 20
    max_dist_to_plane_icp: 0.5
    power_planarity: 2
    threshold_orientation_norm: 0.1
    threshold_translation_norm: 0.01
    debug_print: false
    point_to_plane_with_distortion: false
    distance: CT_POINT_TO_PLANE
    num_closest_neighbors: 1
    beta_location_consistency: 0.001
    beta_constant_velocity: 0.000
    beta_orientation_consistency: 0.000
    beta_small_velocity: 0.01
    solver: CERES
    debug_viz: true
    min_num_residuals: 200
    max_num_residuals: 600
    loss_function: CAUCHY
    ls_max_num_iters: 10
    ls_num_threads: 8
    ls_sigma: 0.2
    ls_tolerant_min_threshold: 0.05
Segmentation fault (core dumped)

HeXu1, 2022-01-13:
and i'm confused with "kitti_raw_transform_trajectory_frame()" function:
R_Tr,T_Tr: does this mean extrinsic lidar to left camera ,T_lidar2came;
center_R,center_t: does this mean lidar absolute pose ,T_lidar2world;
And follows transformation means T_lidar2came*T_lidar2world*inv(T_lidar2came)? it seems not right.
center_R = R_Tr * center_R * R_Tr.transpose();
center_t = -center_R * T_Tr + T_Tr + R_Tr * center_t;

thanks a lot.

HeXu1, 2022-01-13:
T_lidar2came * T_lidar2world * inv(T_lidar2came)

HeXu1, 2022-01-17:
Hi,
 I rebuild it with gcc-9,it works when build with ".\ct_icp_build.sh Release "Unix Makefiles" ON OFF" ,but failed with ".\ct_icp_build.sh Release "Unix Makefiles" ON ON". the output is "Target glfw could not be found". the glfw has installed in "cmake-build-Release/external" path,but cann't link well.
hope for your reply,thanks.

HeXu1, 2022-01-17:
I have fixed it by replace "find_package(glfw3 REQUIRED CONFIG PATHS ${GLFW_DIR})" with "find_package(glfw3 3.3 REQUIRED CONFIG PATHS ${GLFW_DIR})" in external.cmake file.

pdell-kitware, 2022-01-17:
So it didn't work with gcc8 ? That is strange as I had no problem with gcc-7.5. I will investigate.

Your latest problem is probably due to conflicting glfw installed on your system and cmake-build-Release/external

The "proper way" to fix it would be to add NO_DEFAULT_PATH to the cmake command 

find_package(glfw3 REQUIRED CONFIG PATHS ${GLFW_DIR} NO_DEFAULT_PATH)

HeXu1, 2022-01-18:
1) yes, gcc8 didn't work.
2) I build successfully with "find_package(glfw3 REQUIRED CONFIG PATHS ${GLFW_DIR} NO_DEFAULT_PATH)"
3) and the last problem about "center_R = R_Tr * center_R * R_Tr.transpose()", I understand the first "R_Tr" means transformation from lidar odometry to visual odometry . because the two odometry start at the same time.so the extrinsic pram means the two odometry's transformation.
thanks a lot.

HeXu1, 2022-01-18:
And by the way, how to change pcd file to ply used in ct-icp. I found the "writeFile()" function in "PlyFile.h" ,but I cann't use it well. thanks.

closed 2022-01-19 by pdell-kitware