jedeschaud/ct_icp #9: dev Branch `CMake Error: Invalid value used with --config`

MiguelAgueda, 2022-03-07:
Hello, I am attempting to build the `dev` branch to use CT ICP with ROS. 

**OS:** _Ubuntu 18.04_
**CMake:** _V3.23.0_
**Clang:** _10.0_ 
**Branch:** _dev_

I can only get halfway through the steps in [Step 1: Superbuild, SlamCore, ROSCore](https://github.com/jedeschaud/ct_icp/tree/dev#step-1-superbuild-slamcore-roscore) before I get a build error. 

# Attempting to Build #

```bash
myUser@myUsermain:~/Documents/ct_icp/$ mkdir .cmake-build-superbuild && cd .cmake-build-superbuild

myUser@myUsermain:~/Documents/ct_icp/.cmake-build-superbuild$ cmake ../superbuild
-- Configuring done
-- Generating done
-- Build files have been written to: /home/myUser/Documents/ct_icp/.cmake-build-superbuild

myUser@myUsermain:~/Documents/ct_icp/.cmake-build-superbuild$ cmake --build . --config Release -v
/usr/local/bin/cmake -S/home/myUser/Documents/ct_icp/superbuild -B/home/myUser/Documents/ct_icp/.cmake-build-superbuild --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/bin/cmake -E cmake_progress_start /home/myUser/Documents/ct_icp/.cmake-build-superbuild/CMakeFiles /home/myUser/Documents/ct_icp/.cmake-build-superbuild//CMakeFiles/progress.marks
/usr/bin/make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/myUser/Documents/ct_icp/.cmake-build-superbuild'
/usr/bin/make  -f CMakeFiles/MappingResearchKEU_superbuild.dir/build.make CMakeFiles/MappingResearchKEU_superbuild.dir/depend
make[2]: Entering directory '/home/myUser/Documents/ct_icp/.cmake-build-superbuild'
cd /home/myUser/Documents/ct_icp/.cmake-build-superbuild && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/myUser/Documents/ct_icp/superbuild /home/myUser/Documents/ct_icp/superbuild /home/myUser/Documents/ct_icp/.cmake-build-superbuild /home/myUser/Documents/ct_icp/.cmake-build-superbuild /home/myUser/Documents/ct_icp/.cmake-build-superbuild/CMakeFiles/MappingResearchKEU_superbuild.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/myUser/Documents/ct_icp/.cmake-build-superbuild'
/usr/bin/make  -f CMakeFiles/MappingResearchKEU_superbuild.dir/build.make CMakeFiles/MappingResearchKEU_superbuild.dir/build
make[2]: Entering directory '/home/myUser/Documents/ct_icp/.cmake-build-superbuild'
[  6%] Performing update step for 'MappingResearchKEU_superbuild'
cd /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild && /usr/local/bin/cmake -P /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/tmp/MappingResearchKEU_superbuild-gitupdate.cmake
Enter passphrase for key '/home/myUser/.ssh/id_rsa': 
HEAD is now at f666378 [fix]
[ 12%] No patch step for 'MappingResearchKEU_superbuild'
/usr/local/bin/cmake -E echo_append
/usr/local/bin/cmake -E touch /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-patch
[ 18%] Performing configure step for 'MappingResearchKEU_superbuild'
cd /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build && /usr/local/bin/cmake -DSUPERBUILD_INSTALL_DIR=/home/myUser/Documents/ct_icp/superbuild/../install -DWITH_G2O=OFF -DWITH_OPENCV=OFF -DWITH_SQLITE=OFF -DWITH_VIZ3D=OFF "-GUnix Makefiles" /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild
Re-run cmake no build system arguments
INFO [Superbuild] -- [Generation] -- Generating GTest dependency
INFO [Superbuild] -- [Generation] -- Generating GLOG dependency
INFO [Superbuild] -- [Generation] -- Generating Eigen3 dependency
INFO [Superbuild] -- [Generation] -- Generating Ceres dependency
INFO [Superbuild] -- [Generation] -- Generating yaml-cpp dependency
INFO [Superbuild] -- [Generation] -- Generating cereal dependency
INFO [Superbuild] -- [Generation] -- Generating tessil dependency
INFO [Superbuild] -- [Generation] -- Generating nanoflann dependency
INFO [Superbuild] -- [Generation] -- Generating tclap dependency
INFO [Superbuild] -- [Generation] -- Generating colormap dependency
INFO [Superbuild] -- [Generation] -- Generating tinyply dependency
-- Configuring done
-- Generating done
-- Build files have been written to: /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build
cd /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build && /usr/local/bin/cmake -E touch /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-configure
[ 25%] Performing build step for 'MappingResearchKEU_superbuild'
cd /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build && /usr/local/bin/cmake --build /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build --config --target install
CMake Error: Invalid value used with --config
Usage: cmake --build <dir>             [options] [-- [native-options]]
       cmake --build --preset <preset> [options] [-- [native-options]]
Options:
  <dir>          = Project binary directory to be built.
  --preset <preset>, --preset=<preset>
                 = Specify a build preset.
  --list-presets
                 = List available build presets.
  --parallel [<jobs>], -j [<jobs>]
                 = Build in parallel using the given number of jobs. 
                   If <jobs> is omitted the native build tool's 
                   default number is used.
                   The CMAKE_BUILD_PARALLEL_LEVEL environment variable
                   specifies a default parallel level when this option
                   is not given.
  --target <tgt>..., -t <tgt>... 
                 = Build <tgt> instead of default targets.
  --config <cfg> = For multi-configuration tools, choose <cfg>.
  --clean-first  = Build target 'clean' first, then build.
                   (To clean only, use --target 'clean'.)
  --resolve-package-references={on|only|off}
                 = Restore/resolve package references during build.
  --verbose, -v  = Enable verbose output - if supported - including
                   the build commands to be executed. 
  --             = Pass remaining options to the native tool.
CMakeFiles/MappingResearchKEU_superbuild.dir/build.make:85: recipe for target 'MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-build' failed
make[2]: *** [MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-build] Error 1
make[2]: Leaving directory '/home/myUser/Documents/ct_icp/.cmake-build-superbuild'
CMakeFiles/Makefile2:84: recipe for target 'CMakeFiles/MappingResearchKEU_superbuild.dir/all' failed
make[1]: *** [CMakeFiles/MappingResearchKEU_superbuild.dir/all] Error 2
make[1]: Leaving directory '/home/myUser/Documents/ct_icp/.cmake-build-superbuild'
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2
```

# Error # 

After rerunning the CMake in verbose mode, I found the error to be caused by running the following command without specifying the configuration to use. 

```bash
cd /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build && /usr/local/bin/cmake --build /home/myUser/Documents/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build --config --target install
```

## Possible Fix ##

My best guess is that the package will need to be built with the `Release` configuration? 
At the end of the above command, I believe that:
`/src/MappingResearchKEU_superbuild-build --config --target install`
should be replaced with something like 
`/src/MappingResearchKEU_superbuild-build --config Release --target install`

pdell-kitware, 2022-03-08:
Hi, 

Yes, with the file `ct_icp/superbuild/CMakeLists.txt` which assumes that the variable CMAKE_BUILD_TYPE is set, 

You should be able to proceed using the command `cmake ../superbuild -DCMAKE_BUILD_TYPE=Release` instead.

In any cases, this is an issue, I will fix it today.

MiguelAgueda, 2022-03-08:
@pdell-kitware Thank you!

That seems to have moved me to the next steps.

closed 2022-03-08 by MiguelAgueda