jedeschaud/ct_icp #15: Installation error occured!

bigbigpark, 2022-03-30:
In your installation there are 3 steps to install ct-icp code. <br/>

~~~bash
[1] $ .\ct_icp_build.sh Release "Unix Makefiles" ON ON  # Builds the project in "Release" mode, with "Unix Makefiles" cmake generator, with python binding and with the visualization activated
[2] $ source env.sh                                     # Setup the environment (.so locations) 
[3] .\slam -c default_config.yaml                     # Launches the SLAM
~~~
<br/>
I think the symbol '\' need to be changed as '/' because it is about unix system. <br/>
Anyway, when I put the command [1], the errors occur.
<br/>

Error is , <br/>
~~~bash
/home/bigbigpark/ct_icp/src/ct_icp/utils.hpp:5:10: fatal error: filesystem: No such file or directory
 #include <filesystem>
~~~

<br/>
The library <filesystem> is available more than C++17. I checked CMakeLists.txt.... <br/>
How can I build this package ? <br/>

<br/>

Thanks

pdell-kitware, 2022-03-30:
Hi bigbigpark, 

Thanks for your comments, I will update the Readme to clarify, 

Can you specify which compiler you use ? 

You need a c++ compiler which fully supports C++17, 
If you use GCC, you should use a version >= 9, 
Or a clang version >= 8 should be fine, 

If you don't want to change default C++ compiler paths on your system, 

You can specify before invoking `./ct_icp_build.sh ...` the following lines: (this will help CMAKE find the proper compiler)

`export CC=/usr/bin/gcc-9 # path-to-gcc-9 | similarly find clang`
`export CXX=/usr/bin/g++-9 # path-to-g++-9 | clang++` 
`./ct_icp_build.sh Release "Unix Makefiles" ON ON`

Note: You probably need to erase the CMakeCache.txt before calling the script (otherwise CMake will use the cached compiler versions)

bigbigpark, 2022-03-31:
Thanks for fast replying!! <br/>
I found my compiler is gcc-7 so I updated it as gcc-9 <br/>
The build is done with your help! <br/>
<br/>
Plus I need a password at here `https://cloud.mines-paristech.fr/index.php/s/UwgVFtiTOmrgKp5` <br/>
Where can I get the password? <br/>
It said ` (you can find it in an answer in the issues)` but I couldn't find it

bigbigpark, 2022-03-31:
@pdell-kitware  <br/>

I found that in the closed issues! <br/>
Thanks

closed 2022-03-31 by bigbigpark