jedeschaud/ct_icp #17: I have a question about the article "CT-ICP"

bigbigpark, 2022-04-12:
I don't know whether this Github issue pages are right place for question... <br/><br/>

I have read your paper `CT-ICP: Real-time Elastic LiDAR Odometry with Loop Closure`. <br/>
I successfully have built your [ct_icp](https://github.com/jedeschaud/ct_icp) code using KITTI dataset with [viz3d](https://github.com/pierdell/viz3d/tree/viz3d-old) <br/>

Here is my question <br/>
What is exactly the role of [pyLiDAR-SLAM](https://github.com/Kitware/pyLiDAR-SLAM) ? <br/>
The paper says that ct_icp is lidar-only odometry and has loop closure itself using g2o <br/>
Is it enough for SLAM  to use ct_icp ? <br/>
I would appreciate it if you could answer my question...! <br/>

<br/>
Best regards.

pdell-kitware, 2022-04-12:
Hi bigbigpark, 

This is the right place for these questions, thanks for your interest in the project !

So in the paper, we implemented the LoopClosure in pyLidar-SLAM.

Historically, pyLiDAR-SLAM was the main project where I could implement / benchmark SLAM algorithms, 
We made a wrapper for CT-ICP in python and integrated it into pyLidar-SLAM.

Since then, (and really for time management of my PhD thesis), my focus has been on C++ code and ct_icp improvements. 

So we are currently working on an implementation of the loop closure in C++ (with ROS support), as we found that it was more relevant in the short-term for most use-case of SLAM. It will probably appear in another gitlab project, but there will be pointers to it in this project. It should appear in the coming weeks (there is a working version which still needs some polishing).
In the mean time, you will find the LoopClosure implementation we used in our paper in pyLidarSLAM.

However our aim is (when I will have some time) to update the python wrapping to integrate ct-icp with pyLiDAR-SLAM, so we can use the neat parameter grid-search that comes with hydra.

So right now our SLAM (in this project) is really an odometry : ie correct locally, but no compensation for large trajectory errors.

Hope it answers your question, 

Best Regards,

bigbigpark, 2022-04-13:
I totally understood what I didn't know. Thank you for your kind reply. <br/>

Thank you for sharing your wonderful project and giving good inspiration to many people. <br/><br/>

Good luck.

closed 2022-04-13 by bigbigpark