jedeschaud/ct_icp #20: debugging at vscode

D0NGJE, 2022-04-19:
Hi.

I tried to debugging slam.cpp at vscode. 

I got a syntax error like this photo.
[image]

I don't know what is the problem. 

I hope you reply

pdell-kitware, 2022-04-19:
Hi, 

I think you have a problem with the `viz3d`  installation of branch master, 
Have you installed viz3d in the superbuild ? 

Can you describe the installation messages ?

Best regards

D0NGJE, 2022-04-19:
thank you for your reply.

I using debugging mode in vscode. 
[image]

By the way, I successfully built and run it, but when I pressed F5 for debugging in vscode the above error occured .<br/>

[CT-ICP]
The command I input,
~~~bash
$ ./ct_icp_build.sh Release "Unix Makefiles" ON ON 
~~~

[image]

Best regards.

pdell-kitware, 2022-04-20:
So building the SLAM seems to work, 

But you can't compile it on your IDE, is that right ?

Have you tried running the slam similarly to the readme instuctions ? 
Running the command `./slam -c <path-to-config>`

Does this work ?

I am not familiar with vscode, so I can't help you on this problem. But you can try to comment all the code between 
the preprocessor #if CT_ICP_WITH_VIZ see if its compiles

D0NGJE, 2022-04-20:
./slam -c default_config.yaml is work.

Thank you for your reply.

closed 2022-04-20 by D0NGJE