jedeschaud/ct_icp #21: Running ct_icp using another dataset

bigbigpark, 2022-04-20:
Hi, I have recenlty built and run [ct_icp](https://github.com/jedeschaud/ct_icp)  successfully! <br/>
But I have difficulty running another dataset. <br/>
I wonder how ct_icp exactly knows about timestamp between frames. <br/>
While studying your [paper](https://arxiv.org/abs/2109.12979) and code, I found something new. <br/>

Like this [default_config.yaml](https://github.com/jedeschaud/ct_icp/blob/cfb263eaede5a554113687f11428b885269ceceb/default_config.yaml#L15-L16) (permalink highlighted), I found that ct_icp code uses KITTI dataset in the form of `frame_####.ply` format.<br/>
But It cannot work on my own data(*.ply) <br/>

<br/>
When I visualize it on CloudCompare, <br/>
here the **red** one is **frame_0001.ply**, the **black** one is **frame_0005.ply**. <br/>
We can think the ego vehicle moves forward. <br/>

[image]

<br/>

Here, I found something new. <br/>

In CloudCompare I got to know the ply file of KITTI has `x, y, z and timestamp` fields. <br/>

[image]

On the other hand, I have my own data like <br/>

[image]

I think the difference is whether it has a field of `timestamp(PLY_FLOAT32)` <br/>

<br/>

Here is my real question! <br/>
Could you explain the ply format and how ct_icp uses that format briefly if you can. <br/>
Plus, I want to my `.bag`/`.pcd` file to `.ply` which has only `x, y, z and timestamp` file format. <br/>
I couldn't find any solution about that. <br/>

Best regards.

bigbigpark, 2022-04-20:
Thanks for replying <br/>
I use the branch of `master` so I found `PlyFile.cpp` <br/>

<br/>
Could you tell me how I use  your PLYFile parser / reader and convert them to the right format(*.ply: x, y, z and timestamp) for running ct_icp like KITTI ? <br/>
I have a `bag` file including `sensor_msgs/PointCloud2` <br/>

pdell-kitware, 2022-04-21:
You can find an example for the code to parse the PLY file in the
[dataset.cpp](https://github.com/jedeschaud/ct_icp/blob/master/src/ct_icp/dataset.cpp) source file
(See function read_ply_pointcloud)

bigbigpark, 2022-04-24:
Thanks you so much!
I could apply my dataset to ct_icp! 

Interestingly, I have 300 data (frame_0000.ply to frame_0300.ply) from Ouster LiDAR.
All data have x, y, z and timestamp properties. However the timestamp in the all data is set to 0.0.

I thought ct_icp cannot work but it worked well! How do...? I think ct_icp use the begin and end timestamp of the data.

pdell-kitware, 2022-04-24:
That's great ! 

If you can don't hesitate to share screenshots to show us the results !

So if all frames are equal to zero, essentially the interpolated frame during optimization is always the same, 
so that is probably why it still works, note however that the frames are probably not corrected

bigbigpark, 2022-04-26:
Now it's not clear on me but I'll try to understand the structure. <br/>

What is the meaning of **R** and **t** ? <br/>
I know CT_ICP estimates the beginning and end pose at one lidar scan.<br/>
Each pose has `Rotation` and `Translation`, but in the code below it writes a number of R and T. <br/>
Could you explain why the R is 3x3 matrix and T is 3x1 vector? <br/>
[io.cpp](https://github.com/jedeschaud/ct_icp/blob/cfb263eaede5a554113687f11428b885269ceceb/src/ct_icp/io.cpp#L61-L64)

bigbigpark, 2022-04-27:
Oh! I get it ! <br/>
Thank you so much :)

closed 2022-04-27 by bigbigpark