jedeschaud/ct_icp #22: Doubt about CT_ICP_GN function.

Lurvelly, 2022-04-21:
Hi, 
  Thanks for sharing. Your CT_ICP paper is very useful, and the code is beautiful.
  But I'm confused. While studying your paper and code, I found that there are some formula that haven't mentioned in your paper. During the step 4 in the "CT_ICP_GN"(Gauss-Newton solver) function, I haven't understand the vector u(cbx, cby, cbz, nbx, nby, nbz, cex, cey, cez, nex, ney, nez)  represents in your paper.  What's the result of the previous point cross product the closest point normal used for? I'm doubt that I loss some import information.
  Could you please tell me where I can find related paper or math formula?

pdell-kitware, 2022-04-21:
Hi, 

Thanks for your support !

So this is a derivation of the point-to-plane with small angle approximation, 

(you can see the reference paper [here](https://www.comp.nus.edu.sg/~lowkl/publications/lowk_point-to-plane_icp_techrep.pdf) )

Note, however that we have two poses to optimize (instead of one in the mentioned technical report)

Lurvelly, 2022-04-22:
Hi, pdell-kitware

Thanks for your kind reply.

Through the reference paper, I think I already know how it works.

Best Regards.

closed 2022-04-22 by Lurvelly