jedeschaud/ct_icp #23: Question about initializing lidar motion

bigbigpark, 2022-05-04:
Hi again <br/>

I know the initalization of the  lidar motion is one of the important thing in ICP <br/>

In your code , [here](https://github.com/jedeschaud/ct_icp/blob/cfb263eaede5a554113687f11428b885269ceceb/src/ct_icp/odometry.cpp#L209-L221) <br/>

At the beginning pose of current scan, it uses the end pose of previous scan. <br/>
I don't catch up how it inits the end pose of current scan? <br/>

I don't understand above code <br/>
I would appreciate if you can explain that code, if not can you recommend something to refer ? <br/>

Best regards.

pdell-kitware, 2022-05-05:
Hi, 

So we have different modes, (constant velocity or no initialization), 
With constant velocity, it uses the previous relative pose to initialize the motion

With no initialization, it simply uses the previous begin time.

bigbigpark, 2022-05-05:
Thanks a lot ! <br/>

closed 2022-05-05 by bigbigpark