jedeschaud/ct_icp #24: How to get the Derivatives (Jacobians)

YangSiri, 2022-06-16:
Hi, @jedeschaud 

I think you've done a great job in the field of SLAM. The results of CT-ICP is significant. The coding is good too. Would you please tell me how to get the **derivatives (jacobians)** of _ct_plane_norm_residual_ with respect to the corresponding two poses? I see you didn't use any form of cost function defined in ceres such as _AutoDiffCostFunction_. So i wonder how the optimization works by gradients?

https://github.com/jedeschaud/ct_icp/blob/1ba7ce704e9994d39076089ea3fc0dc4d856fe84/src/ct_icp/cost_functions.h#L71

pdell-kitware, 2022-06-20:
Hi @YangSiri, 

So we are indeed actually using the *AutoDiffCostFunction*, 

However in the file you refered to, we only define the Cost Functors, 

The cost functions are defined here: 

https://github.com/jedeschaud/ct_icp/blob/1ba7ce704e9994d39076089ea3fc0dc4d856fe84/src/ct_icp/ct_icp.cpp#L229

Does this answer your question ? 

Best regards,

YangSiri, 2022-06-21:
Thanks, @pdell-kitware 

I see your point. So i wonder whether you analytically infer the derivatives which i'm working on recently. 

Very glad to receive your reply~

chengwei0427, 2023-06-05:
> 

Hi, @YangSiri ,have you analytically infer the derivatives?
I'm working on it recently, and cannot get a good result.