jedeschaud/ct_icp #27: Unable to run with own dataset

SubMishMar, 2022-07-16:
Hello,
I have a folder with *.ply files with the timestamps as filenames. These ply files have been generated from Ouster LiDAR. I am finding it difficult to run this package with my dataset even though I have changed a few things in the default config file like the root folder and the dataset option. I have set the dataset option to PLY_DIRECTORY . 

I get a error which says:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  PLY Directory is not supported by read_pointcloud. See the DirectoryIterator.
Aborted (core dumped)
```
I'll appreciate help in fixing this issue.

pdell-kitware, 2022-07-26:
Hi, following my answer to Issue #29, 

I'll improve the API for the Dataset in order to define your own datasets.

pdell-kitware, 2022-07-27:
Hi, 

I committed a new version of the code, which should simplify your problem, 
You should be able to run the code from the PLY directory, 

If your schema is the different from the one we expect (vertex [x, y, z, timestamp / t]), there will be a bit of work to specify the schema of the point cloud (ie how to read the different properties from the point cloud)

Don't hesitate to let me know If you still have some problem.

Lynda549, 2023-06-23:
Hello,
Write now Im using ouster LiDAR and i don't know how to lunch ct-icpon my Ouster pcap. I would like to understand how to do this.
Thank you;
Cordially.