jedeschaud/ct_icp #29: How to run this on MulRan dataset

nachovizzo, 2022-07-25:
Hello again, thanks for your contribution. I'm trying to run this code on the MulRan dataset without success.

If you ever ran this code on the dataset, can you please provide some pointers? 

I also tried to run with the "PLY_DIRECTORY" with no success:

```sh
WARNING: Logging before InitGoogleLogging() is written to STDERR
I20220725 11:23:17.709126 2555704 slam.cpp:351] Creating directory .outputs/
I20220725 11:23:17.713934 2555704 dataset.cpp:278] Found Sequence PLY_DIR
terminate called after throwing an instance of 'std::runtime_error'
  what():  PLY Directory is not supported by read_pointcloud. See the DirectoryIterator.
[1]    2555704 IOT instruction (core dumped)  ./slam -c default_config.yaml
```

Inspecting the code, I see there is no method to read the `PLY` directory.
https://github.com/jedeschaud/ct_icp/blob/1ba7ce704e9994d39076089ea3fc0dc4d856fe84/src/ct_icp/dataset.cpp#L306

Related to #27, #26, #21

pdell-kitware, 2022-07-26:
Hi, 

Thanks for your interest ! 

I assume that your are using the `master` branch ? 

We made quite a few changes in the `dev` branch (and we are actually still intensely on it). 

I will make a cleaner API for the dataset so that you can define your own loader. I'll point to the branch in this issue.

nachovizzo, 2022-07-26:
Hi @pdell-kitware that would be amazing! Thanks for the fast reply!!!. I didn't know about the `dev` branch. I will be waiting  for your instructions then :) I guess this issue is also related to #28

pdell-kitware, 2022-07-27:
Hi, 

So as mentionned in the Readme, you can create your own datasets, and use the files `odometry_runner.h, odometry_runner.cpp` to build a script which will launch on the sequences you defined. 

When I have a bit more time integrate the MulRAN dataset in the code, but I will be quite busy in the coming weeks, so in the meantime you can simply define your own sequences. 

Don't hesitate to ask if you have any problem.

closed 2022-11-08 by nachovizzo