jedeschaud/ct_icp #32: Installation Problems for new release

pdell-kitware, 2022-07-28:


nachovizzo, 2022-07-28:
Thanks for opening this issue, the first error on the first translation unit it starts like
```
Consolidate compiler generated dependencies of target SlamCore
[  1%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/types.cxx.o
In file included from /home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/types.h:15,
                 from /home/ivizzo/dev/slam/ct_icp/src/SlamCore/types.cxx:1:
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h:311:61: error: ‘std::optional’ has not been declared
  311 |                                                             std::optional<std::vector<std::string>> properties = {}) {
      |                                                             ^~~
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h:311:74: error: expected ‘,’ or ‘...’ before ‘<’ token
  311 |                                                             std::optional<std::vector<std::string>> properties = {}) {
      |                                                                          ^
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h:346:85: error: ‘std::optional’ has not been declared
  346 |                                                                                     std::optional<std::vector<std::string>> properties) { \
      |                                                                                     ^~~
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h:364:5: note: in expansion of macro ‘SCHEMA_EIGEN_MATRIX_SPECIALIZATION’
  364 |     SCHEMA_EIGEN_MATRIX_SPECIALIZATION(float, 3, 1)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h:346:98: error: expected ‘,’ or ‘...’ before ‘<’ token
  346 |                                                                                     std::optional<std::vector<std::string>> properties) { \
      |                                                                                                  ^
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h:364:5: note: in expansion of macro ‘SCHEMA_EIGEN_MATRIX_SPECIALIZATION’
  364 |     SCHEMA_EIGEN_MATRIX_SPECIALIZATION(float, 3, 1)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/types.h:15,
                 from /home/ivizzo/dev/slam/ct_icp/src/SlamCore/types.cxx:1:
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h: In function ‘slam::ItemSchema::Builder slam::BuilderFromSingleElementData(const string&, int) [with DataT = Eigen::Matrix<float, 3, 1>; std::string = std::__cxx11::basic_string<char>]’:
/home/ivizzo/dev/slam/ct_icp/src/SlamCore/../../include/SlamCore/data/schema.h:352:13: error: ‘properties’ was not declared in this scope; did you mean ‘PropertySize’?
  352 |         if (properties) { \
      |             ^~~~~~~~~~
```

pdell-kitware, 2022-07-28:
Ok, thanks ! So it seems that std::optional is not found. strange. 

Can you try adding the line 
```
#include <optional>
```

At the top of the schema.h file ?

nachovizzo, 2022-07-28:
So moving forward, I included the header files that were missing (I recommend checking out "include-what-you-use"). And now it did compile, but we are getting linking errors.

## Patch to compile it
```diff
diff --git a/include/SlamCore/data/schema.h b/include/SlamCore/data/schema.h
index 9b135c0..0cc7287 100644
--- a/include/SlamCore/data/schema.h
+++ b/include/SlamCore/data/schema.h
@@ -2,6 +2,7 @@
 #define SlamCore_SCHEMA_H

 #include <map>
+#include <optional>

 #include <Eigen/Dense>
 #include <glog/logging.h>
diff --git a/include/SlamCore/timer.h b/include/SlamCore/timer.h
index 9deebfa..2740dcf 100644
--- a/include/SlamCore/timer.h
+++ b/include/SlamCore/timer.h
@@ -4,6 +4,7 @@
 #include <string>
 #include <map>
 #include <chrono>
+#include <stdexcept>

 namespace slam {

diff --git a/src/SlamCore/utils.cxx b/src/SlamCore/utils.cxx
index 293414b..4b346f4 100644
--- a/src/SlamCore/utils.cxx
+++ b/src/SlamCore/utils.cxx
@@ -115,7 +115,7 @@ void _posix_signal_handler(int sig, siginfo_t *siginfo, void *context) {
     _Exit(1);
 }
```

## New linking errors

```sh
Consolidate compiler generated dependencies of target SlamCore
[  1%] Linking CXX shared library libSlamCore.so
/usr/bin/ld: CMakeFiles/SlamCore.dir/eval.cxx.o:(.bss+0x0): multiple definition of `slam::pose_less_by_fid_and_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/eval.cxx.o:(.bss+0x1): multiple definition of `slam::pose_less_by_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x1): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/eval.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/io.cxx.o:(.bss+0x0): multiple definition of `slam::pose_less_by_fid_and_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/io.cxx.o:(.bss+0x1): multiple definition of `slam::pose_less_by_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x1): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/io.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/imu.cxx.o:(.bss+0x0): multiple definition of `slam::pose_less_by_fid_and_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/imu.cxx.o:(.bss+0x1): multiple definition of `slam::pose_less_by_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x1): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/imu.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/reactors/pointcloud_writer.cxx.o:(.bss+0x0): multiple definition of `slam::pose_less_by_fid_and_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/reactors/pointcloud_writer.cxx.o:(.bss+0x1): multiple definition of `slam::pose_less_by_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x1): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/reactors/pointcloud_writer.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/experimental/synthetic.cxx.o:(.bss+0x0): multiple definition of `slam::pose_less_by_fid_and_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/experimental/synthetic.cxx.o:(.bss+0x1): multiple definition of `slam::pose_less_by_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x1): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/experimental/synthetic.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/experimental/map.cxx.o:(.bss+0x0): multiple definition of `slam::pose_less_by_fid_and_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/experimental/map.cxx.o:(.bss+0x1): multiple definition of `slam::pose_less_by_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x1): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/experimental/map.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/algorithm/grid_sampling.cxx.o:(.bss+0x0): multiple definition of `slam::pose_less_by_fid_and_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/algorithm/grid_sampling.cxx.o:(.bss+0x1): multiple definition of `slam::pose_less_by_timestamp'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x1): first defined here
/usr/bin/ld: CMakeFiles/SlamCore.dir/algorithm/grid_sampling.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
[... cut ...]
/usr/bin/ld: CMakeFiles/SlamCore.dir/pointcloud.cxx.o:(.bss+0x2): multiple definition of `slam::pose_less_by_fid'; CMakeFiles/SlamCore.dir/trajectory.cxx.o:(.bss+0x2): first defined here
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/SlamCore/CMakeFiles/SlamCore.dir/build.make:408: src/SlamCore/libSlamCore.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:307: src/SlamCore/CMakeFiles/SlamCore.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

## Compiler

I'm using gcc on ubuntu. The superbuild did work

pdell-kitware, 2022-07-28:
Thanks I'll check this out, 

Can you tell me which version of gcc do you have ? With gcc < 7.5 we had some compilation problem (because we use C++-17)

nachovizzo, 2022-07-28:
```sh
$ gcc --version
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

pdell-kitware, 2022-07-28:
Thanks, I'll try to replicate the bug, and I'll get back to you

nachovizzo, 2022-07-28:
Amazing! Thanks a lot for the help!!!!

pdell-kitware, 2022-07-28:
If you try a different compiler, it would be very helpful, 

For example if you have `clang` on your system, before running the cmake install script, 
You can export the environment variables:

```
which clang                     # Checks the clang installation

# -- Setup environment variables
export C=clang               
export CXX=clang++

# -- CT_ICP Install
mkdir cmake-build-release && cd  cmake-build-release                  #< Create the build directory
cmake .. -DCMAKE_BUILD_TYPE=Release                                   #< (2) Configure with the desired options (specify arguments with -D<arg_name>=<arg_value>)
cmake --build . --target install --config Release --parallel 12       #< Build and Install the project

```

(And do delete the cmake cache before launch the new installation)

pdell-kitware, 2022-07-28:
Ok I replicated the bug, I'll notify you when I have the fix available

pdell-kitware, 2022-07-28:
This should now be resolved with commit [f2db6a1](https://github.com/jedeschaud/ct_icp/commit/f2db6a1064d953c5950170bf4ddf7da29500f3ef)

nachovizzo, 2022-07-28:
Yay! Finally it's building, now let's see how to run this on the data :) 

Thanks!

nachovizzo, 2022-07-28:
You can close the issue

nachovizzo, 2022-07-28:
Sorry to bother again, I still can't run this code

```sh
ct_icp/install/CT_ICP/bin (master) ./run_odometry -c ../../../config/odometry/driving_config.yaml
./run_odometry: error while loading shared libraries: libglog.so.0: cannot open shared object file: No such file or directory
```

Which magic export I might be missing?

nachovizzo, 2022-07-28:
Update, most libraries are missing from the binary:

```sh
libSlamCore.so
        libyaml-cpp.so.0.6 => not found
        libceres.so.2 => not found
        libtinyply.so => not found
        libglog.so.0 => not found
```

pdell-kitware, 2022-07-28:
Can you run `readelf -d run_odometry` ? 
And notably look at the `RUNPATH` ? 

Sorry to use you as an early tester 😅

nachovizzo, 2022-07-28:
```
adelf -d run_odometry

Dynamic section at offset 0x5b7f0 contains 35 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libCT_ICP.so]
 0x0000000000000001 (NEEDED)             Shared library: [libSlamCore.so]
 0x0000000000000001 (NEEDED)             Shared library: [libglog.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libyaml-cpp.so.0.6]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000001d (RUNPATH)            Library runpath: [/home/ivizzo/dev/slam/ct_icp/install/CT_ICP/lib]
 0x000000000000000c (INIT)               0x15000
 0x000000000000000d (FINI)               0x4e0a4
 0x0000000000000019 (INIT_ARRAY)         0x5b440
 0x000000000000001b (INIT_ARRAYSZ)       24 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x5b458
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x3b0
 0x0000000000000005 (STRTAB)             0x59c0
 0x0000000000000006 (SYMTAB)             0x1640
 0x000000000000000a (STRSZ)              43715 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000003 (PLTGOT)             0x5ca60
 0x0000000000000002 (PLTRELSZ)           3984 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x13610
 0x0000000000000007 (RELA)               0x10b98
 0x0000000000000008 (RELASZ)             10872 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000000000001e (FLAGS)              BIND_NOW
 0x000000006ffffffb (FLAGS_1)            Flags: NOW PIE
 0x000000006ffffffe (VERNEED)            0x10a28
 0x000000006fffffff (VERNEEDNUM)         4
 0x000000006ffffff0 (VERSYM)             0x10484
 0x000000006ffffff9 (RELACOUNT)          336
 0x0000000000000000 (NULL)               0x0
```

```
 echo $RUNPATH
```
Is empty

pdell-kitware, 2022-07-28:
I managed to replicate the error, I'll notify you when it is solved

ajxdhe, 2022-07-29:
Hello I got the error in **cmake --build . --target install --config Release --parallel 12** 
the detail is

Consolidate compiler generated dependencies of target SlamCore
[  0%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o
In file included from /home/skywalker/ct_icp/src/SlamCore/io.cxx:6:0:
/home/skywalker/ct_icp/src/SlamCore/io.cxx: In function ‘std::vector<slam::PLYSchemaMapper::PLYProperty> slam::{anonymous}::DefaultPropertiesFromSchema(const std::vector<slam::ItemSchema>&, bool)’:
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:30:32: error: ‘filesystem’ is not a member of ‘std’
 #define __SLAM_FILENAME__ std::filesystem::path(__FILE__).filename()
                                ^
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:32:53: note: in expansion of macro ‘__SLAM_FILENAME__’
 #define SLAM_LOCATION_INFO ("[File: " + std::string(__SLAM_FILENAME__)  + ", Line: " + std::to_string(__LINE__) + "] -- ")
                                                     ^~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:34:45: note: in expansion of macro ‘SLAM_LOCATION_INFO’
 #define SLAM_LOG(CATEGORY) LOG(CATEGORY) << SLAM_LOCATION_INFO
                                             ^~~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/io.cxx:994:25: note: in expansion of macro ‘SLAM_LOG’
                         SLAM_LOG(WARNING) << "The element `" << ply_element_name << "` appears twice in the schema";
                         ^~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:30:32: note: suggested alternative:
 #define __SLAM_FILENAME__ std::filesystem::path(__FILE__).filename()
                                ^
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:32:53: note: in expansion of macro ‘__SLAM_FILENAME__’
 #define SLAM_LOCATION_INFO ("[File: " + std::string(__SLAM_FILENAME__)  + ", Line: " + std::to_string(__LINE__) + "] -- ")
                                                     ^~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:34:45: note: in expansion of macro ‘SLAM_LOCATION_INFO’
 #define SLAM_LOG(CATEGORY) LOG(CATEGORY) << SLAM_LOCATION_INFO
                                             ^~~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/io.cxx:994:25: note: in expansion of macro ‘SLAM_LOG’
                         SLAM_LOG(WARNING) << "The element `" << ply_element_name << "` appears twice in the schema";
                         ^~~~~~~~
In file included from /usr/include/c++/7/experimental/filesystem:36:0,
                 from /home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:18,
                 from /home/skywalker/ct_icp/src/SlamCore/io.cxx:6:
/usr/include/c++/7/experimental/bits/fs_fwd.h:45:11: note:   ‘std::experimental::filesystem’
 namespace filesystem
           ^~~~~~~~~~
src/SlamCore/CMakeFiles/SlamCore.dir/build.make:187: recipe for target 'src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o' failed
make[2]: *** [src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o] Error 1
CMakeFiles/Makefile2:306: recipe for target 'src/SlamCore/CMakeFiles/SlamCore.dir/all' failed
make[1]: *** [src/SlamCore/CMakeFiles/SlamCore.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

ajxdhe, 2022-07-29:
I tried this method [https://github.com/jedeschaud/ct_icp/issues/32#issuecomment-1198163415](url)

but got the new error

`cmake --build . --target install --config Release --parallel 12  
[  1%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/types.cxx.o
[  2%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/geometry.cxx.o
[  4%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/config_utils.cxx.o
[  4%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/trajectory.cxx.o
[  4%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/ceres_utils.cxx.o
[  5%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/utils.cxx.o
[  5%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o
[  6%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/imu.cxx.o
[  7%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/eval.cxx.o
[  8%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/reactors/pointcloud_writer.cxx.o
[  9%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/timer.cxx.o
[ 10%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/reactors/scheduler.cxx.o
[ 10%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/experimental/synthetic.cxx.o
[ 11%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/experimental/map.cxx.o
[ 12%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/experimental/neighborhood.cxx.o
[ 13%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/algorithm/grid_sampling.cxx.o
/home/skywalker/ct_icp/src/SlamCore/io.cxx:994:25: error: no member named 'filesystem' in namespace 'std'; did you mean 'std::experimental::filesystem'?
                        SLAM_LOG(WARNING) << "The element `" << ply_element_name << "` appears twice in the schema";
                        ^~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:34:45: note: expanded from macro 'SLAM_LOG'
#define SLAM_LOG(CATEGORY) LOG(CATEGORY) << SLAM_LOCATION_INFO
                                            ^~~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:32:53: note: expanded from macro 'SLAM_LOCATION_INFO'
#define SLAM_LOCATION_INFO ("[File: " + std::string(__SLAM_FILENAME__)  + ", Line: " + std::to_string(__LINE__) + "] -- ")
                                                    ^~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:30:27: note: expanded from macro '__SLAM_FILENAME__'
#define __SLAM_FILENAME__ std::filesystem::path(__FILE__).filename()
                          ^~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/experimental/filesystem:47:11: note: 'std::experimental::filesystem' declared here
namespace filesystem
          ^
1 error generated.
[ 14%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/data/buffer_collection.cxx.o
src/SlamCore/CMakeFiles/SlamCore.dir/build.make:187: recipe for target 'src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o' failed
make[2]: *** [src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o] Error 1
make[2]: *** 正在等待未完成的任务....
`

ajxdhe, 2022-07-29:
I tried this method to install 
which clang                     # Checks the clang installation

# -- Setup environment variables
export C=clang               
export CXX=clang++

# -- CT_ICP Install
mkdir cmake-build-release && cd  cmake-build-release                  #< Create the build directory
cmake .. -DCMAKE_BUILD_TYPE=Release                                   #< (2) Configure with the desired options (specify arguments with -D<arg_name>=<arg_value>)
cmake --build . --target install --config Release --parallel 12       #< Build and Install the project

but got the new error
 cmake --build . --target install --config Release --parallel 12  
[  1%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/types.cxx.o
[  2%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/geometry.cxx.o
[  4%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/config_utils.cxx.o
[  4%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/trajectory.cxx.o
[  4%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/ceres_utils.cxx.o
[  5%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/utils.cxx.o
[  5%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o
[  6%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/imu.cxx.o
[  7%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/eval.cxx.o
[  8%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/reactors/pointcloud_writer.cxx.o
[  9%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/timer.cxx.o
[ 10%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/reactors/scheduler.cxx.o
[ 10%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/experimental/synthetic.cxx.o
[ 11%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/experimental/map.cxx.o
[ 12%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/experimental/neighborhood.cxx.o
[ 13%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/algorithm/grid_sampling.cxx.o
/home/skywalker/ct_icp/src/SlamCore/io.cxx:994:25: error: no member named 'filesystem' in namespace 'std'; did you mean 'std::experimental::filesystem'?
                        SLAM_LOG(WARNING) << "The element `" << ply_element_name << "` appears twice in the schema";
                        ^~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:34:45: note: expanded from macro 'SLAM_LOG'
#define SLAM_LOG(CATEGORY) LOG(CATEGORY) << SLAM_LOCATION_INFO
                                            ^~~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:32:53: note: expanded from macro 'SLAM_LOCATION_INFO'
#define SLAM_LOCATION_INFO ("[File: " + std::string(__SLAM_FILENAME__)  + ", Line: " + std::to_string(__LINE__) + "] -- ")
                                                    ^~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/src/SlamCore/../../include/SlamCore/utils.h:30:27: note: expanded from macro '__SLAM_FILENAME__'
#define __SLAM_FILENAME__ std::filesystem::path(__FILE__).filename()
                          ^~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/experimental/filesystem:47:11: note: 'std::experimental::filesystem' declared here
namespace filesystem
          ^
1 error generated.
[ 14%] Building CXX object src/SlamCore/CMakeFiles/SlamCore.dir/data/buffer_collection.cxx.o
src/SlamCore/CMakeFiles/SlamCore.dir/build.make:187: recipe for target 'src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o' failed
make[2]: *** [src/SlamCore/CMakeFiles/SlamCore.dir/io.cxx.o] Error 1
make[2]: *** 正在等待未完成的任务....

pdell-kitware, 2022-07-29:
@nachovizzo 

I fixed the issue (setting the path to the external shared libraries in the `RPATH` of the executables and libraries built by the project)

The tests run on the CI (see this [JOB](https://github.com/jedeschaud/ct_icp/runs/7577432909?check_suite_focus=true))

So you should be good to go, but let me know if you have any more problems

pdell-kitware, 2022-07-29:
@ajxdhe, 

The problem is that you are compiling again in gcc 7.5 which does not support completely `<filesystem>`, 

As can be seen in:
```
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/experimental/filesystem:47:11: note: 'std::experimental::filesystem' declared here
namespace filesystem
```

Your problem (in your second answer) is that you probably did not delete the CMakeCache.txt before configuring the project again. 

Try deleting the `cmake-build-release` directory, then 

```
export CC=clang
export CXX=clang++

mkdir cmake-build-release && cd  cmake-build-release                  #< Create the build directory
cmake .. -DCMAKE_BUILD_TYPE=Release                                   #< (2) Configure with the desired options (specify arguments with -D<arg_name>=<arg_value>), add -DWITH_VIZ3D=ON to install with the GUI
cmake --build . --target install --config Release --parallel 12       #< Build and Install the project
```

If you have any more issues on this topic, please write them on Issue #35

nachovizzo, 2022-07-29:
> @nachovizzo
>
> I fixed the issue (setting the path to the external shared libraries in the `RPATH` of the executables and libraries built by the project)
>
> The tests run on the CI (see this [...]

I confirm this solve the build error, now I have another errors :angel:

pdell-kitware, 2022-07-29:
That's not possible, the tests pass 😛

Thanks again for all the effort, it was very helpful !

pdell-kitware, 2022-07-29:
I'll close this for now .

closed 2022-07-29 by pdell-kitware