jedeschaud/ct_icp #35: Compiler Issues

pdell-kitware, 2022-07-29:
CT-ICP is written in  C++-17, 

This means you need a compiler which support C++17, 

To configure the compiler you want before building the project, write the following lines:

```
export CC=<path-to-c-compiler>
export CXX=<path-to-cxx-compiler>

cmake .. etc..
```

closed 2022-07-29 by pdell-kitware

ajxdhe, 2022-07-30:
Thank you for your answer
but I got the new issue in ROS
when I
cd ros/roscore
mkdir cmake-build-release && cd  cmake-build-release                  #< Create the build directory
cmake .. -DCMAKE_BUILD_TYPE=Release      
It got the error
cmake .. -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
INFO [ROSCore] -- Variable "SUPERBUILD_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install
INFO [ROSCore] -- Variable "CT_ICP_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install/CT_ICP/lib/cmake.
CMake Error at /home/skywalker/ct_icp/cmake/superbuild_utils.cmake:27 (include):
  include could not find requested file:

    /home/skywalker/install/superbuild_import.cmake
Call Stack (most recent call first):
  CMakeLists.txt:33 (SLAM_INCLUDE_SUPERBUILD)

CMake Error at CMakeLists.txt:36 (find_package):
  Could not find a package configuration file provided by "SlamCore" with any
  of the following names:

    SlamCoreConfig.cmake
    slamcore-config.cmake

  Add the installation prefix of "SlamCore" to CMAKE_PREFIX_PATH or set
  "SlamCore_DIR" to a directory containing one of the above files.  If
  "SlamCore" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/skywalker/ct_icp/ros/roscore/cmake-build-release/CMakeFiles/CMakeOutput.log".

pdell-kitware, 2022-07-30:
Hi @ajxdhe, please see the answer in Issue #37