jedeschaud/ct_icp #37: ROS Installation Issue

pdell-kitware, 2022-07-30:
Thank you for your answer
but I got the new issue in ROS
when I
cd ros/roscore
mkdir cmake-build-release && cd  cmake-build-release                  #< Create the build directory
cmake .. -DCMAKE_BUILD_TYPE=Release      
It got the error
cmake .. -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
INFO [ROSCore] -- Variable "SUPERBUILD_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install
INFO [ROSCore] -- Variable "CT_ICP_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install/CT_ICP/lib/cmake.
CMake Error at /home/skywalker/ct_icp/cmake/superbuild_utils.cmake:27 (include):
  include could not find requested file:

    /home/skywalker/install/superbuild_import.cmake
Call Stack (most recent call first):
  CMakeLists.txt:33 (SLAM_INCLUDE_SUPERBUILD)

CMake Error at CMakeLists.txt:36 (find_package):
  Could not find a package configuration file provided by "SlamCore" with any
  of the following names:

    SlamCoreConfig.cmake
    slamcore-config.cmake

  Add the installation prefix of "SlamCore" to CMAKE_PREFIX_PATH or set
  "SlamCore_DIR" to a directory containing one of the above files.  If
  "SlamCore" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/skywalker/ct_icp/ros/roscore/cmake-build-release/CMakeFiles/CMakeOutput.log".

_Originally posted by @ajxdhe in https://github.com/jedeschaud/ct_icp/issues/35#issuecomment-1200143070_

pdell-kitware, 2022-07-30:
Have you performed the superbuild steps (STEP 1 and 2 ?), before this step ?

The fact that `/home/skywalker/install/superbuild_import.cmake` can't be found suggests otherwise. 
Could you tell me where your `Superbuild` and `SlamCore, CT_ICP` installation were performed ?

ajxdhe, 2022-07-30:
I performed the step1 ( add -DWITH_VIZ3D=ON)
but got the error
home/skywalker/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/viz3d/src/viz3d-build/_deps/imgui-src/backends/imgui_impl_glfw.cpp: In function ‘void ImGui_ImplGlfw_UpdateMonitors()’:
/home/skywalker/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/viz3d/src/viz3d-build/_deps/imgui-src/backends/imgui_impl_glfw.cpp:502:9: error: ‘glfwGetMonitorWorkarea’ was not declared in this scope
         glfwGetMonitorWorkarea(glfw_monitors[n], &x, &y, &w, &h);
         ^~~~~~~~~~~~~~~~~~~~~~
/home/skywalker/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/viz3d/src/viz3d-build/_deps/imgui-src/backends/imgui_impl_glfw.cpp:502:9: note: suggested alternative: ‘glfwGetMonitorName’
         glfwGetMonitorWorkarea(glfw_monitors[n], &x, &y, &w, &h);
         ^~~~~~~~~~~~~~~~~~~~~~
         glfwGetMonitorName
CMakeFiles/imgui.dir/build.make:215: recipe for target 'CMakeFiles/imgui.dir/_deps/imgui-src/backends/imgui_impl_glfw.cpp.o' failed
make[8]: *** [CMakeFiles/imgui.dir/_deps/imgui-src/backends/imgui_impl_glfw.cpp.o] Error 1
CMakeFiles/Makefile2:134: recipe for target 'CMakeFiles/imgui.dir/all' failed
make[7]: *** [CMakeFiles/imgui.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make[6]: *** [all] Error 2
CMakeFiles/viz3d.dir/build.make:85: recipe for target 'viz3d/src/viz3d-stamp/viz3d-build' failed
make[5]: *** [viz3d/src/viz3d-stamp/viz3d-build] Error 2
CMakeFiles/Makefile2:315: recipe for target 'CMakeFiles/viz3d.dir/all' failed
make[4]: *** [CMakeFiles/viz3d.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/MappingResearchKEU_superbuild.dir/build.make:85: recipe for target 'MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-build' failed
make[2]: *** [MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-build] Error 2
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/MappingResearchKEU_superbuild.dir/all' failed
make[1]: *** [CMakeFiles/MappingResearchKEU_superbuild.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

ajxdhe, 2022-07-30:
the gcc version is 11

ajxdhe, 2022-07-30:
I perform again

pdell-kitware, 2022-07-30:
Is this the first error that you have ?

Also, 
Can you try after installing all the dependencies installed in the CI ? 

``` 
-y libsdl2-dev gcc-11-multilib g++-11-multilib libfreetype6-dev libvulkan-dev freeglut3-dev libxrandr-dev libglfw3-dev
```

ajxdhe, 2022-07-30:
> Have you performed the superbuild steps (STEP 1 and 2 ?), before this step ?
> 
> The fact that `/home/skywalker/install/superbuild_import.cmake` can't be found suggests otherwise. Could you tell me where your `Superbuild` and `SlamCore, CT_ICP` installation were performed ?

do you mean I have to add  -DWITH_VIZ3D=ON in step1 or 2?

pdell-kitware, 2022-07-30:
You need to add it to the two **configure** steps of cmake

First for the superbuild

```cmake ../superbuild -DCMAKE_BUILD_TYPE=Release -DWITH_VIZ3D=ON```

Then for the main build:
```cmake .. -DCMAKE_BUILD_TYPE=Release -DWITH_VIZ3D=ON```

ajxdhe, 2022-07-30:
I superbuilded  ,Builded and installed CT-ICP library  again but the error also exist

cd ros/roscore
mkdir cmake-build-release && cd cmake-build-release #< Create the build directory
cmake .. -DCMAKE_BUILD_TYPE=Release
It got the error
cmake .. -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
INFO [ROSCore] -- Variable "SUPERBUILD_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install
INFO [ROSCore] -- Variable "CT_ICP_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install/CT_ICP/lib/cmake.
CMake Error at /home/skywalker/ct_icp/cmake/superbuild_utils.cmake:27 (include):
include could not find requested file:

/home/skywalker/install/superbuild_import.cmake

Call Stack (most recent call first):
CMakeLists.txt:33 (SLAM_INCLUDE_SUPERBUILD)

CMake Error at CMakeLists.txt:36 (find_package):
Could not find a package configuration file provided by "SlamCore" with any
of the following names:

SlamCoreConfig.cmake
slamcore-config.cmake

Add the installation prefix of "SlamCore" to CMAKE_PREFIX_PATH or set
"SlamCore_DIR" to a directory containing one of the above files. If
"SlamCore" provides a separate development package or SDK, be sure it has
been installed.

-- Configuring incomplete, errors occurred!
See also "/home/skywalker/ct_icp/ros/roscore/cmake-build-release/CMakeFiles/CMakeOutput.log".

Originally posted by @ajxdhe in https://github.com/jedeschaud/ct_icp/issues/35#issuecomment-1200143070

pdell-kitware, 2022-07-30:
Ok, 

1. Did you build with with -DWITH_VIZ3D=ON or not ?

2. Can you print the files of the installation directory ? 

`ls -R  /home/skywalker/install/CT_ICP`

3. Can you give me your the hash of your current version of the code ?

ajxdhe, 2022-07-30:
1. I did not build with -DWITH_VIZ3D=ON
2. I print it and i shows
 ls -R CT_ICP/
CT_ICP/:
bin  include  lib

CT_ICP/bin:
run_odometry

CT_ICP/include:
ct_icp  ct_icp-viz3d  SlamCore  SlamCore-viz3d

CT_ICP/include/ct_icp:
algorithm         ct_icp.h   map.h                    odometry.h  utils.h
config.h          dataset.h  motion_model.h           reactors
cost_functions.h  io.h       neighborhood_strategy.h  types.h

CT_ICP/include/ct_icp/algorithm:
sampling.h

CT_ICP/include/ct_icp/reactors:
dataset_loader.h  logger.h  registration.h

CT_ICP/include/ct_icp-viz3d:
dataset_window.h  hilti_dataset_window.h  viz3d_utils.h

CT_ICP/include/SlamCore:
algorithm       conversion.h     geometry.h    reactors      utils.h
cereal.h        data             imu.h         timer.h
ceres_utils.h   eval.h           io.h          traits.h
concurrent      experimental     pointcloud.h  trajectory.h
config_utils.h  generic_tools.h  predicates.h  types.h

CT_ICP/include/SlamCore/algorithm:
grid_sampling.h

CT_ICP/include/SlamCore/concurrent:
blocking_queue.h

CT_ICP/include/SlamCore/data:
buffer_collection.h  buffer.h  proxy_ref.h  schema.h  view.h

CT_ICP/include/SlamCore/experimental:
iterator  map.h  neighborhood.h  synthetic.h

CT_ICP/include/SlamCore/experimental/iterator:
base_iterator.h   proxy_view_iterator.h  view_iterator.h
proxy_iterator.h  transform_iterator.h

CT_ICP/include/SlamCore/reactors:
handler.h  notifier.h  observer.h  pointcloud_writer.h  reactor.h  scheduler.h

CT_ICP/include/SlamCore-viz3d:
viz3d_utils.h  viz3d_windows.h

CT_ICP/lib:
cmake  libCT_ICP.so  libSlamCore.so

CT_ICP/lib/cmake:
CT_ICPConfig.cmake   CT_ICPTargets-release.cmake  SlamCoreTargets.cmake
CT_ICPTargets.cmake  SlamCoreConfig.cmake         SlamCoreTargets-release.cmake

3. the hash of my current version of the code

commit 884b21b4e1c2205dac7835328a29a9edccf52841 (HEAD -> master, origin/master, origin/HEAD)
Author: pdell-kitware <59437242+pdell-kitware@users.noreply.github.com>
Date:   Thu Jul 28 17:03:24 2022 +0200

    Update readme.md

diff --git a/readme.md b/readme.md
index 43cc071..df7da0b 100644
--- a/readme.md
+++ b/readme.md
@@ -56,8 +56,10 @@ Superbuild**](https://gitlab.kitware.com/keu-computervision/MappingResearchKEU/S
 ```bash
 mkdir .cmake-build-superbuild && cd .cmake-build-superbuild     #< Creates the cmake folder
 cmake ../superbuild                                             #< (1) Configure step 
-cmake --build . --config Release                                #< Build step (Downloads and install the dependencies)
+cmake --build . --config Release                                #< Build step (Downloads and install the dependencies), add -DWITH_VIZ3D=ON to install with the GUI
:

commit 884b21b4e1c2205dac7835328a29a9edccf52841 (HEAD -> master, origin/master, origin/HEAD)
Author: pdell-kitware <59437242+pdell-kitware@users.noreply.github.com>
Date:   Thu Jul 28 17:03:24 2022 +0200

    Update readme.md

diff --git a/readme.md b/readme.md
index 43cc071..df7da0b 100644
--- a/readme.md
+++ b/readme.md
@@ -56,8 +56,10 @@ Superbuild**](https://gitlab.kitware.com/keu-computervision/MappingResearchKEU/S
 ```bash
 mkdir .cmake-build-superbuild && cd .cmake-build-superbuild     #< Creates the cmake folder
 cmake ../superbuild                                             #< (1) Configure step 
-cmake --build . --config Release                                #< Build step (Downloads and install the dependencies)
+cmake --build . --config Release                                #< Build step (Downloads and install the dependencies), add -DWITH_VIZ3D=ON to install with the GUI

pdell-kitware, 2022-07-30:
Can you try pulling the latest version of `master` ?

ajxdhe, 2022-07-30:
what should I do ?
forgive me I am a just freshman in university

pdell-kitware, 2022-07-30:
`git pull origin master`

ajxdhe, 2022-07-30:
> `git pull origin master`

I finished it what should i do next step

pdell-kitware, 2022-07-30:
You can try to rebuild (not the superbuild, you can skip step 0), 

**Step 2** + **Step 3**,

ajxdhe, 2022-07-30:
the step 3 also have the error but it seem different
when I  cmake .. -DCMAKE_BUILD_TYPE=Release
INFO [ROSCore] -- Variable "SUPERBUILD_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install
INFO [ROSCore] -- Variable "CT_ICP_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/install/CT_ICP/lib/cmake.
CMake Error at /home/skywalker/ct_icp/cmake/superbuild_utils.cmake:27 (include):
  include could not find requested file:

    /home/skywalker/install/superbuild_import.cmake
Call Stack (most recent call first):
  CMakeLists.txt:33 (SLAM_INCLUDE_SUPERBUILD)

CMake Error at /home/skywalker/ct_icp/cmake/superbuild_utils.cmake:34 (message):
   [SlamCore-cmake] -- Could not find target Eigen3::Eigen
Call Stack (most recent call first):
  /home/skywalker/ct_icp/cmake/external.cmake:8 (SLAM_CHECK_TARGETS)
  CMakeLists.txt:34 (include)

-- Configuring incomplete, errors occurred!
See also "/home/skywalker/ct_icp/ros/roscore/cmake-build-release/CMakeFiles/CMakeOutput.log".

pdell-kitware, 2022-07-30:
Is there a file  `/home/skywalker/install/superbuild_import.cmake` ?

ajxdhe, 2022-07-30:
no It isn't

ajxdhe, 2022-07-30:
the file is in /home/skywalker/ct_icp/install/

pdell-kitware, 2022-07-30:
Ok thanks I'll look into that !

pdell-kitware, 2022-07-30:
Can you try the branch `fix_#37` ? 

`git pull origin fix_#37`

Then only **step 3**

ajxdhe, 2022-07-30:
When I do  cmake --build . --target install --config Release --parallel 12   it got the error
~/ct_icp/ros/roscore/cmake-build-release$ cmake --build . --target install --config Release --parallel 12  
[ 50%] Building CXX object src/CMakeFiles/ROSCore.dir/pc2_conversion.cxx.o
[ 33%] Building CXX object src/CMakeFiles/ROSCore.dir/nav_msgs_conversion.cxx.o
[ 50%] Building CXX object src/CMakeFiles/ROSCore.dir/pcl_conversion.cxx.o
[ 66%] Linking CXX shared library libROSCore.so
[ 66%] Built target ROSCore
[ 83%] Building CXX object test/unit/CMakeFiles/test_pc2_conversion.dir/test_pc2_conversion.cxx.o
[100%] Linking CXX executable test_pc2_conversion
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/skywalker/ct_icp/ros/roscore/cmake-build-release/src/libROSCore.so)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/skywalker/ct_icp/install/glog/lib/libglog.so.0)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/skywalker/ct_icp/install/glog/lib/libglog.so.0)
CMake Error at /usr/share/cmake-3.23/Modules/GoogleTestAddTests.cmake:112 (message):
  Error running test executable.

    Path: '/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion'
    Result: 1
    Output:
      

Call Stack (most recent call first):
  /usr/share/cmake-3.23/Modules/GoogleTestAddTests.cmake:225 (gtest_discover_tests_impl)

test/unit/CMakeFiles/test_pc2_conversion.dir/build.make:597: recipe for target 'test/unit/test_pc2_conversion' failed
make[2]: *** [test/unit/test_pc2_conversion] Error 1
make[2]: *** 正在删除文件“test/unit/test_pc2_conversion”
CMakeFiles/Makefile2:2661: recipe for target 'test/unit/CMakeFiles/test_pc2_conversion.dir/all' failed
make[1]: *** [test/unit/CMakeFiles/test_pc2_conversion.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

pdell-kitware, 2022-07-30:
Have you run the line: 

`sudo apt-get update -y libsdl2-dev gcc-11-multilib g++-11-multilib libfreetype6-dev libvulkan-dev freeglut3-dev libxrandr-dev libglfw3-dev` ?

pdell-kitware, 2022-07-30:
Aren't you compiling with gcc ? 

What is the compiler for this execution, is it `clang` or `gcc` ?

If possible try with clang before **step 3** (delete the cache), 
And do again:
`
export CC=clang
export CXX=clang++
`

ajxdhe, 2022-07-30:
the clang version is 9.0
the gcc is 11
first time I use gcc 11
the second time I use clang 9.0
the error is

[ 16%] Building CXX object src/CMakeFiles/ROSCore.dir/nav_msgs_conversion.cxx.o
[ 33%] Building CXX object src/CMakeFiles/ROSCore.dir/pc2_conversion.cxx.o
[ 50%] Building CXX object src/CMakeFiles/ROSCore.dir/pcl_conversion.cxx.o
[ 66%] Linking CXX shared library libROSCore.so
[ 66%] Built target ROSCore
[ 83%] Building CXX object test/unit/CMakeFiles/test_pc2_conversion.dir/test_pc2_conversion.cxx.o
[100%] Linking CXX executable test_pc2_conversion
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_M_split_cmpts()@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::status(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_List(std::filesystem::__cxx11::path::_List const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::__throw_bad_array_new_length()@GLIBCXX_3.4.29’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::parent_path() const@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/glog/lib/libglog.so.0.5.0：对‘std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::create_directories(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_List()@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::operator/=(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_Impl_deleter::operator()(std::filesystem::__cxx11::path::_List::_Impl*) const@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26’未定义的引用
clang: error: linker command failed with exit code 1 (use -v to see invocation)
test/unit/CMakeFiles/test_pc2_conversion.dir/build.make:597: recipe for target 'test/unit/test_pc2_conversion' failed
make[2]: *** [test/unit/test_pc2_conversion] Error 1
CMakeFiles/Makefile2:2661: recipe for target 'test/unit/CMakeFiles/test_pc2_conversion.dir/all' failed
make[1]: *** [test/unit/CMakeFiles/test_pc2_conversion.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

pdell-kitware, 2022-07-30:
I'll try to replicate the bug with the CI, and I'll get back to you

ajxdhe, 2022-07-30:
I sincerely appreciate  your work .I should have a sleep （It's midnight in China）qwq

ajxdhe, 2022-07-31:
should I try the branch fix?

pdell-kitware, 2022-07-31:
Ok, 

So I managed to build ROSCore on `github action`, 

Can you: 

1. Tell me what is your platform (Ubuntu 18, Debian, etc ...), if you have installed ROS, and if so which version ? 
2. Clean your installation folder for the roscore build
3. Checkout master `git checkout master` at the root of the project
4. Build ct_icp (*Step 2*) and give me the complete LOG of the console

5. **Instead of Step 3**, do (if you do not have `melodic` installed replace `melodic` by your ros distro):

```
          source /opt/ros/melodic/setup.bash
          cd ros/roscore
          mkdir cmake-build-release && cd  cmake-build-release
          cmake .. -DCMAKE_BUILD_TYPE=Release
          cmake --build . --target install --config Release --parallel 12
```

And give me the result of each execution (separately)

ajxdhe, 2022-07-31:
1.I use the ubuntu 18.04 and ros melodic

3.the result of the git checkout master
 git checkout master
已经位于 'master'
您的分支领先 'origin/master' 共 1 个提交。
  （使用 "git push" 来发布您的本地提交）
(Translation:git checkout master

Already in'Master'

Your branch is 1 submission ahead of'origin/master'.

(use "git push" to publish your local submission))

4 the complete LOG of the console is 

skywalker@skywalker-ASUS-TUF-Gaming-F15-FX506HCB-FX506HCB:~/ct_icp/cmake-build-release$ cmake .. -DCMAKE_BUILD_TYPE=Release   
INFO [CT_ICP] -- Variable "SUPERBUILD_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/ct_icp/install
INFO [CT_ICP] -- Variable "SLAMCORE_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/ct_icp/install/SlamCore
INFO [CT_ICP] -- Variable "CT_ICP_INSTALL_DIR" not specified. Trying to set it to /home/skywalker/ct_icp/install/CT_ICP
INFO [Superbuild] -- Successfully found target glog::glog
INFO [Superbuild] --  Variable EIGEN_DIR not found, setting it to directory /home/skywalker/ct_icp/install/Eigen3/share/eigen3/cmake
INFO [Superbuild] -- Successfully Found Target Eigen3::Eigen
-- Found required Ceres dependency: Eigen version 3.3.7 in /home/skywalker/ct_icp/install/Eigen3/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 2.0.0 installed in: /home/skywalker/ct_icp/install/Ceres with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, Multithreading]
INFO [Superbuild] -- Successfully found target Ceres::ceres
INFO [Superbuild] -- Successfully found target yaml-cpp
INFO [Superbuild] -- Successfully found target GTest::gtest
INFO [Superbuild] -- Successfully found target cereal
INFO [Superbuild] -- Successfully found target tclap::tclap
INFO [Superbuild] -- Successfully found target tsl::robin_map
INFO [Superbuild] -- Successfully found target nanoflann::nanoflann
INFO [Superbuild] -- Successfully found target colormap::colormap
INFO [Superbuild] -- Successfully found target tinyply::tinyply
INFO /home/skywalker/ct_icp/install/Ceres/lib/libceres.so.2.0.0
INFO /home/skywalker/ct_icp/install/glog/lib/libglog.so.0.5.0
INFO /home/skywalker/ct_icp/install/yaml-cpp/lib/libyaml-cpp.so.0.6.3
INFO /home/skywalker/ct_icp/install/tinyply/lib/libtinyply.so
INFO -- [CT-ICP] -- Appending to the INSTALL RPATH the RPATH to the external libraries: 
		[:/home/skywalker/ct_icp/install/Ceres/lib:/home/skywalker/ct_icp/install/glog/lib:/home/skywalker/ct_icp/install/yaml-cpp/lib:/home/skywalker/ct_icp/install/tinyply/lib]
INFO [CT_ICP] -- "WITH_GTSAM=OFF gtsam dependent targets will not be built"
-- Configuring done
-- Generating done
-- Build files have been written to: /home/skywalker/ct_icp/cmake-build-release
skywalker@skywalker-ASUS-TUF-Gaming-F15-FX506HCB-FX506HCB:~/ct_icp/cmake-build-release$ cmake --build . --target install --config Release --parallel 12    
Consolidate compiler generated dependencies of target SlamCore
[ 17%] Built target SlamCore
Consolidate compiler generated dependencies of target test_timer
Consolidate compiler generated dependencies of target test_types
Consolidate compiler generated dependencies of target test_trajectory
Consolidate compiler generated dependencies of target test_eval
Consolidate compiler generated dependencies of target test_io
Consolidate compiler generated dependencies of target test_yaml_utils
Consolidate compiler generated dependencies of target test_pointcloud
Consolidate compiler generated dependencies of target test_geometry
Consolidate compiler generated dependencies of target test_buffer_collection
Consolidate compiler generated dependencies of target test_buffer
Consolidate compiler generated dependencies of target test_config
[ 20%] Built target test_timer
[ 20%] Built target test_pointcloud
[ 21%] Built target test_buffer_collection
[ 25%] Built target test_geometry
[ 25%] Built target test_trajectory
[ 29%] Built target test_eval
[ 29%] Built target test_io
[ 30%] Built target test_config
[ 32%] Built target test_types
[... 218 lines cut ...]
[ 83%] Building CXX object test/unit/CMakeFiles/test_pc2_conversion.dir/test_pc2_conversion.cxx.o
[100%] Linking CXX executable test_pc2_conversion
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/skywalker/ct_icp/ros/roscore/cmake-build-release/src/libROSCore.so)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/skywalker/ct_icp/install/glog/lib/libglog.so.0)
/home/skywalker/ct_icp/ros/roscore/cmake-build-release/test/unit/test_pc2_conversion: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/skywalker/ct_icp/install/glog/lib/libglog.so.0)
[... cut ...]
  /usr/share/cmake-3.23/Modules/GoogleTestAddTests.cmake:225 (gtest_discover_tests_impl)

test/unit/CMakeFiles/test_pc2_conversion.dir/build.make:597: recipe for target 'test/unit/test_pc2_conversion' failed
make[2]: *** [test/unit/test_pc2_conversion] Error 1
make[2]: *** 正在删除文件“test/unit/test_pc2_conversion”
CMakeFiles/Makefile2:2661: recipe for target 'test/unit/CMakeFiles/test_pc2_conversion.dir/all' failed
make[1]: *** [test/unit/CMakeFiles/test_pc2_conversion.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

pdell-kitware, 2022-07-31:
Ok, thanks, 

Now:
0. `sudo apt-get upgrade libstdc++6` (following [this link](https://stackoverflow.com/questions/44773296/libstdc-so-6-version-glibcxx-3-4-20-not-found))

1. `git pull origin master`, 
2. (Delete the ROSCore build directory *ie* )

3. Give me the first 
```
          export CC=clang
          export CXX=clang++
          source /opt/ros/melodic/setup.bash
          cd ros/roscore
          mkdir cmake-build-release && cd  cmake-build-release
          cmake .. -DCMAKE_BUILD_TYPE=Release
```
Give me the *complete console output* of the previous line, 

```
          cmake --build . --target install --config Release --parallel 12
```

pdell-kitware, 2022-07-31:
**If the above fails:**

4. Have you done this ? `sudo apt-get install libsdl2-dev gcc-11-multilib g++-11-multilib libfreetype6-dev libvulkan-dev freeglut3-dev libxrandr-dev libglfw3-dev`

5. Can you print the output of `strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX`

ajxdhe, 2022-08-01:
_the complete console output of the previous line,_

  cmake --build . --target install --config Release --parallel 12
[ 16%] Building CXX object src/CMakeFiles/ROSCore.dir/nav_msgs_conversion.cxx.o
[ 33%] Building CXX object src/CMakeFiles/ROSCore.dir/pc2_conversion.cxx.o
[ 50%] Building CXX object src/CMakeFiles/ROSCore.dir/pcl_conversion.cxx.o
[ 66%] Linking CXX shared library libROSCore.so
[ 66%] Built target ROSCore
[ 83%] Building CXX object test/unit/CMakeFiles/test_pc2_conversion.dir/test_pc2_conversion.cxx.o
[100%] Linking CXX executable test_pc2_conversion
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_M_split_cmpts()@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::status(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_List(std::filesystem::__cxx11::path::_List const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::__throw_bad_array_new_length()@GLIBCXX_3.4.29’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::parent_path() const@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/glog/lib/libglog.so.0.5.0：对‘std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::create_directories(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_List()@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::operator/=(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_Impl_deleter::operator()(std::filesystem::__cxx11::path::_List::_Impl*) const@GLIBCXX_3.4.26’未定义的引用
/home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26’未定义的引用
clang: error: linker command failed with exit code 1 (use -v to see invocation)
test/unit/CMakeFiles/test_pc2_conversion.dir/build.make:597: recipe for target 'test/unit/test_pc2_conversion' failed
make[2]: *** [test/unit/test_pc2_conversion] Error 1
CMakeFiles/Makefile2:2661: recipe for target 'test/unit/CMakeFiles/test_pc2_conversion.dir/all' failed
make[1]: *** [test/unit/CMakeFiles/test_pc2_conversion.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

_4 I have install the dependencies_

_5.the output of strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX_
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_DEBUG_MESSAGE_LENGTH

imurfutur3, 2023-03-11:
> _the complete console output of the previous line,_
> 
> cmake --build . --target install --config Release --parallel 12 [ 16%] Building CXX object src/CMakeFiles/ROSCore.dir/nav_msgs_conversion.cxx.o [ 33%] Building CXX object src/CMakeFiles/ROSCore.dir/pc2_conversion.cxx.o [ 50%] Building CXX object src/CMakeFiles/ROSCore.dir/pcl_conversion.cxx.o [ 66%] Linking CXX shared library libROSCore.so [ 66%] Built target ROSCore [ 83%] Building CXX object test/unit/CMakeFiles/test_pc2_conversion.dir/test_pc2_conversion.cxx.o [100%] Linking CXX executable test_pc2_conversion /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_M_split_cmpts()@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::status(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_List(std::filesystem::__cxx11::path::_List const&)@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::__throw_bad_array_new_length()@GLIBCXX_3.4.29’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::parent_path() const@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/glog/lib/libglog.so.0.5.0：对‘std::__cxx11::basic_ostringstream<char, std::char_traits, std::allocator >::basic_ostringstream()@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::create_directories(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_List()@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::operator/=(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::filesystem::__cxx11::path::_List::_Impl_deleter::operator()(std::filesystem::__cxx11::path::_List::_Impl*) const@GLIBCXX_3.4.26’未定义的引用 /home/skywalker/ct_icp/install/CT_ICP/lib/libSlamCore.so：对‘std::__cxx11::basic_stringstream<char, std::char_traits, std::allocator >::basic_stringstream()@GLIBCXX_3.4.26’未定义的引用 clang: error: linker command failed with exit code 1 (use -v to see invocation) test/unit/CMakeFiles/test_pc2_conversion.dir/build.make:597: recipe for target 'test/unit/test_pc2_conversion' failed make[2]: *** [test/unit/test_pc2_conversion] Error 1 CMakeFiles/Makefile2:2661: recipe for target 'test/unit/CMakeFiles/test_pc2_conversion.dir/all' failed make[1]: *** [test/unit/CMakeFiles/test_pc2_conversion.dir/all] Error 2 Makefile:145: recipe for target 'all' failed make: *** [all] Error 2
> 
> _4 I have install the dependencies_
> 
> _5.the output of strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX_ GLIBCXX_3.4 GLIBCXX_3.4.1 GLIBCXX_3.4.2 GLIBCXX_3.4.3 GLIBCXX_3.4.4 GLIBCXX_3.4.5 GLIBCXX_3.4.6 GLIBCXX_3.4.7 GLIBCXX_3.4.8 GLIBCXX_3.4.9 GLIBCXX_3.4.10 GLIBCXX_3.4.11 GLIBCXX_3.4.12 GLIBCXX_3.4.13 GLIBCXX_3.4.14 GLIBCXX_3.4.15 GLIBCXX_3.4.16 GLIBCXX_3.4.17 GLIBCXX_3.4.18 GLIBCXX_3.4.19 GLIBCXX_3.4.20 GLIBCXX_3.4.21 GLIBCXX_3.4.22 GLIBCXX_3.4.23 GLIBCXX_3.4.24 GLIBCXX_3.4.25 GLIBCXX_DEBUG_MESSAGE_LENGTH

1. it seems you are in the conda env,  pls try conda deactivate, and /opt/ros/noetic/setup.bash. 
2. you should use higher version cmake, e.g. cmake 3.26. you can ref to this blog <https://blog.csdn.net/heroacool/article/details/129461720?spm=1001.2014.3001.5501>