jedeschaud/ct_icp #40: Superbuild issue

liangyongshi, 2022-08-05:
when I superbuild i met this issue

cmake --build . --config Release 
Scanning dependencies of target MappingResearchKEU_superbuild
[ 12%] Creating directories for 'MappingResearchKEU_superbuild'
[ 25%] Performing download step (git clone) for 'MappingResearchKEU_superbuild'
Cloning into 'MappingResearchKEU_superbuild'...
Already on 'master'
Your branch is up to date with 'origin/master'.
[ 37%] No patch step for 'MappingResearchKEU_superbuild'
[ 50%] Performing update step for 'MappingResearchKEU_superbuild'
Current branch master is up to date.
[ 62%] Performing configure step for 'MappingResearchKEU_superbuild'
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
INFO [Superbuild] -- [Generation] -- Generating GTest dependency
INFO [Superbuild] -- [Generation] -- Generating GLOG dependency
INFO [Superbuild] -- [Generation] -- Generating Eigen3 dependency
INFO [Superbuild] -- [Generation] -- Generating Ceres dependency
INFO [Superbuild] -- [Generation] -- Generating yaml-cpp dependency
INFO [Superbuild] -- [Generation] -- Generating cereal dependency
INFO [Superbuild] -- [Generation] -- Generating tessil dependency
INFO [Superbuild] -- [Generation] -- Generating nanoflann dependency
INFO [Superbuild] -- [Generation] -- Generating tclap dependency
CMake Error at CMakeLists.txt:527 (add_library):
  add_library INTERFACE library requires no source arguments.

CMake Error at CMakeLists.txt:528 (target_include_directories):
  Cannot specify include directories for target "tclap" which is not built by
  this project.

CMake Error at CMakeLists.txt:468 (install):
  install TARGETS given target "tclap" which does not exist.
Call Stack (most recent call first):
  CMakeLists.txt:531 (SUPERBUILD_INSTALL_TARGET)

INFO [Superbuild] -- [Generation] -- Generating colormap dependency
INFO [Superbuild] -- [Generation] -- Generating tinyply dependency
-- Configuring incomplete, errors occurred!
See also "/home/user/rosProject/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/CMakeFiles/CMakeOutput.log".
make[2]: *** [CMakeFiles/MappingResearchKEU_superbuild.dir/build.make:107: MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/MappingResearchKEU_superbuild.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
 
My ubuntu is 20.04

pdell-kitware, 2022-08-05:
Hi, 

Can you tell me what branch are you using (`master` ?), and your `cmake` version ?

qqstar-423, 2022-08-11:
> Hi,
> 
> Can you tell me what branch are you using (`master` ?), and your `cmake` version ?

I had the same problem,im using master,and my cmake version is 3.16.3

qqstar-423, 2022-08-11:
I set the WITH_TCLAP to 0, and the super build is successful.(Superbuild/CMakeLists.txt:507)

pdell-kitware, 2022-08-11:
@qqstar-423, that is probably not a good idea, as you will need TCLAP to build the CT-ICP Library. 

**Short answer:** I think a more recent cmake version would make the problem go away (I use cmake version 3.23)

Other possibility is to change the CMakeLists.txt of the Superbuild which was locally downloaded (normally at `.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild` but it will require a bit of cmake knowledge) (the problem I think is that `add_library(tclap INTERFACE ${TCLAP_HEADERS})` raises a cmake error for earlier version of cmake). 

I will investigate this, thanks for pointing it out, I'll notify the issue when I have replicated and fixed the issue

pdell-kitware, 2022-08-11:
This should now be fixed with the latest fix in the Superbuild

pdell-kitware, 2022-08-11:
No worries, thanks for posting the issue !

Closing this thread

closed 2022-08-11 by pdell-kitware