jedeschaud/ct_icp #44: Launch File Errors

linwk20, 2022-08-19:
Hi Thanks for your sharing! This is a good work.

I am now trying run ct_icp on my own dataset using Ubuntu1804/ROS/Melodic, However, when I roslaunch [ct_icp_on_rosbag.launch](https://github.com/jedeschaud/ct_icp/blob/master/ros/catkin_ws/ct_icp_odometry/launch/ct_icp_on_rosbag.launch), I got warning that Unknown <arg> attribute "required" and Unknown <include> attribute "args", it seems that this two tab has no such attributes?

I wonder have you try this launch file on ros melodic? How can I fix it? Thanks!

pdell-kitware, 2022-08-22:
Hi ! 

So despite these warnings can you run it on your dataset or does it fail ? 
The warning should not be an issue (I have the same), 

Don't forget that you need to specify the following parameters:
 - `rosbag`
 - `topic` points to the point cloud topic. 

If you have failures, can you print the console log with the relevant error messages ?

linwk20, 2022-08-22:
Yes. The program run well with warning, thanks for reply!
Also, I have modified the launch to eliminate the warning.

closed 2022-08-22 by linwk20