jedeschaud/ct_icp #45: Is loop closure enable?

linwk20, 2022-08-19:
Hi Thanks for your great work!

When I trying roslaunch ct_icp_odometry ct_icp_odometry_node, I found it only output odom, not path, so I guess it does not contains loop closure, I wonder how can I enable loop closure in ros?

Thanks!

pdell-kitware, 2022-08-22:
Hi !

So for now there is not yet loop closure available (we are working on it, but we are pretty busy at the moment ^^), 

It should arrive in late september at the earliest on this project.

For the paper we used the loop closure from **pyLIDAR-SLAM**

linwk20, 2022-08-22:
Hi, Thanks for reply.
So did you just run ct_icp and pySLAM offline seperately to get the loop-closure result?
I will try it, thanks!