jedeschaud/ct_icp #46: Termination: NO_CONVERGENCE on my own dataset

linwk20, 2022-08-23:
Hi Thanks for your sharing of this excellent work!

I have ssuccessfully run ct icp on my own dataset, however, I have two question, and I hope you can help me!

(1) My dataset is collected using OS1-128 Lidar, shoud I modified any configuration in the code to adapt to the lidar?

(2) The terminal says : Termination: NO_CONVERGENCE, Is it normal? Or did I mistake something? I have post one output below, the loss is about 0.1.

Thanks for your reading! 

/* ------------------------------------------------------------------------ */
REGISTRATION OF FRAME number 704 (Fid:704) with CERES solver
Number of points in sub-sampled frame: 6292
Initial ego-motion distance: 0.219946
Ceres Solver Report: Iterations: 2, Initial cost: 1.823143e-01, Final cost: 1.734975e-01, Termination: NO_CONVERGENCE
[CT-ICP]: Rotation diff: 0.25545(deg)
[CT-ICP]: Translation diff: 0.0268903(m)
Ceres Solver Report: Iterations: 2, Initial cost: 1.753664e-01, Final cost: 1.741726e-01, Termination: NO_CONVERGENCE
[CT-ICP]: Rotation diff: 0.126741(deg)
[CT-ICP]: Translation diff: 0.0156516(m)
Ceres Solver Report: Iterations: 2, Initial cost: 1.658873e-01, Final cost: 1.649816e-01, Termination: NO_CONVERGENCE
[CT-ICP]: Rotation diff: 0.064464(deg)
[CT-ICP]: Translation diff: 0.0102195(m)
Ceres Solver Report: Iterations: 2, Initial cost: 1.587869e-01, Final cost: 1.583586e-01, Termination: NO_CONVERGENCE
[CT-ICP]: Rotation diff: 0.0440787(deg)
[CT-ICP]: Translation diff: 0.00744236(m)
Ceres Solver Report: Iterations: 2, Initial cost: 1.574686e-01, Final cost: 1.567192e-01, Termination: NO_CONVERGENCE
[CT-ICP]: Rotation diff: 0.0759043(deg)
[CT-ICP]: Translation diff: 0.00731901(m)
[CT-ICP]: Correction (Optim-Init[end])=0.0200416(m)
[CT-ICP]: Correction (Optim-Init[begin])=0.0105974(m)
[CT-ICP]: Begin Pose:
 -0.372328   0.927865  0.0209494    9.44271
 -0.927987  -0.372542 0.00730528    54.7062
 0.0145828 -0.0167208   0.999754  -0.100985
         0          0          0          1
[CT-ICP]: End Pose:
 -0.371037   0.928334  0.0229463    9.35357
 -0.928504  -0.371265 0.00648589    54.5304
 0.0145402 -0.0188992   0.999716 -0.0981214
         0          0          0          1
Trajectory correction [begin(t) - end(t-1)]: 0
Final ego-motion distance: 0.197156
Elapsed Time: 63.3284 (ms)
Inserting points in the Map:  Skipped Frames [0 / 5]  Total Insertions: 704
Cumulative Orientation: 0.117865°
Cumulative Distance: 0.18717m
Ego Orientation: 0.147157°

 [RUNNING]  Bag Time: 1656034744.789619   Duration: 70.598759 / 114.797567               
[CT-ICP] Logged Values:
 -- icp_duration_neighborhood: 32.8843
 -- icp_duration_solve: 21.011
 -- icp_num_iters: 5
 -- icp_total_duration: 55.9733
 -- odometry_duration_sampling: 0.164952
 -- odometry_initialization: 0.720628
 -- odometry_initialization(ms): 6.28906
 -- odometry_map_update(ms): 37.0534
 -- odometry_num_keypoints: 1045
 -- odometry_total: 104.93
 -- odometry_total_duration(ms): 63.3284
 -- odometry_transform(ms): 3.43219
 -- odometry_try_register: 57.0291

 [RUNNING]  Bag Time: 1656034744.839663   Duration: 70.648802 / 114.797567               
/* ------------------------------------------------------------------------ */

pdell-kitware, 2022-08-26:
Hi again !

So, the NO_CONVERGENCE should not necessarily be a problem.
We log the ceres optimizer output, so this just means that we did not wait the ceres problem to converge. 

Indeed, we limit the number of iterations of the optimizer during the ICP, in order to speed up computations. 
This is really a speed tradeof, but can be modified in the parameters (see `CTICPOptions::ls_max_iters` parameter)
Same for the number of iterations of the ICP (which is probably set to 5 at maximum). 

For the optimal parameters of the OS-128, it depends what is your goal accuracy or speed ? 
For accuracy, you would want to augment the number of iterations of ct_icp and of the solver, and the number of keypoints. 
For speed, what you really want is to reduce the number of keypoints and of iterations of ICP. 

I am curious about your results, does it work well with the default parameters or do you have failures ? 
(If you want to post a screenshot of your results it would be greatly appreciated ;) )

Note that we do a sampling of the frame and of keypoints in this frame, which should help with the massive data from the os-128.

linwk20, 2022-08-26:
Hi. Thanks for your detailed reply!
The result did not failed in my test, but is worse than A-LOAM, so that why I post the issue. The evaluation is the revisit error on my own dataset.
According to your reply, I will try to tuned the algorithm for accuracy, and perform further comparison with A-LOAM, Thanks!

pdell-kitware, 2022-08-26:
For precision you should use the config `config/nclt_config.yaml` as a starting point

`/!\` 
Note that the default config is really specialized for driving scenarios, and will not work at all for indoor.

linwk20, 2022-08-26:
Indeed, my dataset is collected in a underground indoor parking lot.
Thanks, I will try tuning.

linwk20, 2022-08-28:
I have tried the OS-128 config and tuned keypoints num and voxel size.
The ICP converged, but the result seems not become better.
Do you have any advice for tuning? 

Thanks!