jedeschaud/ct_icp #47: Cmake Error in Ubuntu20.04

lingbo-yu, 2022-09-01:
>  $ cmake .. -DCMAKE_BUILD_TYPE=Release -DSUPERBUILD_INSTALL_DIR=/home/lingbo/open-source-project/ct_icp/install -DSLAMCORE_INSTALL_DIR=/home/lingbo/open-source-project/ct_icp/install/SlamCore -DCT_ICP_INSTALL_DIR=/home/lingbo/open-source-project/ct_icp/install/CT_ICP -DEIGEN_DIR=/home/lingbo/open-source-project/ct_icp/install/Eigen3/share/eigen3/cmake
> INFO [Superbuild] -- Successfully found target glog::glog
> INFO [Superbuild] -- Successfully Found Target Eigen3::Eigen
> -- Found required Ceres dependency: Eigen version 3.3.7 in /home/lingbo/open-source-project/ct_icp/install/Eigen3/include/eigen3
> -- Found required Ceres dependency: glog
> -- Found required Ceres dependency: gflags
> -- Found Ceres version: 2.0.0 installed in: /home/lingbo/open-source-project/ct_icp/install/Ceres with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, Multithreading]
> INFO [Superbuild] -- Successfully found target Ceres::ceres
> INFO [Superbuild] -- Successfully found target yaml-cpp
> INFO [Superbuild] -- Successfully found target GTest::gtest
> INFO [Superbuild] -- Successfully found target cereal
> INFO [Superbuild] -- Successfully found target tclap::tclap
> INFO [Superbuild] -- Successfully found target tsl::robin_map
> INFO [Superbuild] -- Successfully found target nanoflann::nanoflann
> INFO [Superbuild] -- Successfully found target colormap::colormap
> INFO [Superbuild] -- Successfully found target tinyply::tinyply
> CMake Error at cmake/external.cmake:13 (get_target_property):
>   INTERFACE_LIBRARY targets may only have whitelisted properties.  The
>   property "IMPORTED_RELEASE_LOCATION" is not allowed.
> Call Stack (most recent call first):
>   CMakeLists.txt:51 (include)
> 
> 
> INFO Eigen3::Eigen NOTFOUND
> INFO /home/lingbo/open-source-project/ct_icp/install/Ceres/lib/libceres.so.2.0.0
> INFO /home/lingbo/open-source-project/ct_icp/install/glog/lib/libglog.so.0.5.0
> CMake Error at cmake/external.cmake:13 (get_target_property):
>   INTERFACE_LIBRARY targets may only have whitelisted properties.  The
>   property "IMPORTED_RELEASE_LOCATION" is not allowed.
> Call Stack (most recent call first):
>   CMakeLists.txt:51 (include)
> 
> 
> INFO tsl::robin_map NOTFOUND
> INFO /home/lingbo/open-source-project/ct_icp/install/yaml-cpp/lib/libyaml-cpp.so.0.6.3
> CMake Error at cmake/external.cmake:13 (get_target_property):
>   INTERFACE_LIBRARY targets may only have whitelisted properties.  The
>   property "IMPORTED_RELEASE_LOCATION" is not allowed.
> Call Stack (most recent call first):
>   CMakeLists.txt:51 (include)
> 
> 
> INFO colormap::colormap NOTFOUND
> INFO /home/lingbo/open-source-project/ct_icp/install/tinyply/lib/libtinyply.so
> INFO -- [CT-ICP] -- Appending to the INSTALL RPATH the RPATH to the external libraries: 
> 		[:/home/lingbo/open-source-project/ct_icp/install/Ceres/lib:/home/lingbo/open-source-project/ct_icp/install/glog/lib:/home/lingbo/open-source-project/ct_icp/install/yaml-cpp/lib:/home/lingbo/open-source-project/ct_icp/install/tinyply/lib]
> INFO [CT_ICP] -- "WITH_GTSAM=OFF gtsam dependent targets will not be built"
> -- Configuring incomplete, errors occurred!
> See also "/home/lingbo/open-source-project/ct_icp/cmake-build-release/CMakeFiles/CMakeOutput.log".
> See also "/home/lingbo/open-source-project/ct_icp/cmake-build-release/CMakeFiles/CMakeError.log".

lingbo-yu, 2022-09-01:
it seems eigen3/tsl::robin_map/colormap did not install seccussfully, but the superbuild succeeded. Those deps are already in my install directory: 

> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:15 cereal
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:16 Ceres
> -rw-rw-r-- 1 lingbo lingbo  20K Sep  1 11:18 CMakeCache.txt
> drwxrwxr-x 5 lingbo lingbo 4.0K Sep  1 11:15 CMakeFiles
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:16 colormap
> drwxrwxr-x 3 lingbo lingbo 4.0K Sep  1 11:15 command
> -rw-rw-r-- 1 lingbo lingbo 1.1K Sep  1 11:15 CT_ICPConfig.cmake
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:15 Eigen3
> drwxrwxr-x 2 lingbo lingbo 4.0K Sep  1 10:20 glad
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:20 glfw
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:15 glog
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:14 googletest
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:16 nanoflann
> -rw-rw-r-- 1 lingbo lingbo 1.1K Sep  1 11:15 SlamCoreConfig.cmake
> drwxrwxr-x 5 lingbo lingbo 4.0K Sep  1 11:15 src
> -rw-rw-r-- 1 lingbo lingbo 6.9K Sep  1 10:32 superbuild_import.cmake
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:16 tclap
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:14 tessil
> drwxrwxr-x 6 lingbo lingbo 4.0K Sep  1 11:15 test
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:16 tinyply
> drwxrwxr-x 6 lingbo lingbo 4.0K Sep  1 10:28 vtk
> drwxrwxr-x 4 lingbo lingbo 4.0K Sep  1 10:14 yaml-cpp

owl-d, 2022-09-13:
same Error in Ubuntu 18.04

pdell-kitware, 2022-09-13:
Hi @lingbo-yu, @owl-d, 

Sorry for the delay, can you tell me what is your cmake version ?

owl-d, 2022-09-14:
I used cmake 3.14.0

pelliun, 2022-09-15:
same issue at cmake 3.16.3.
I update cmake version 3.20.5, compile success.

pdell-kitware, 2022-09-15:
@owl-d, 

Either you could upgrade your cmake version, 

Or try the branch pdell-kitware-patch-1 ?

owl-d, 2022-09-21:
cmake 3.20.5 works! Thank you !

closed 2022-12-07 by pdell-kitware