jedeschaud/ct_icp #50: Pybind issue

Unknownnnnnnn, 2022-10-27:
Thank you very much for your work!
I didn't find any bindings to python in the compile command you provided, am I missing something?

pdell-kitware, 2022-11-02:
Hi, 

The python bindings were available in a previous version, 
And we have yet to reimplement them,

To be honest, it wasn't very high our long list of priorities, as we didn't expect them to be much used. 

But if you are interested, I can start working on them again (this should be available next week)

Let me know, 

Cheers

dandric, 2022-11-10:
I'm interested to have python bindings for the current version too and it would help with related work in [pyLiDAR-SLAM](https://github.com/Kitware/pyLiDAR-SLAM) too. Please let me know what are the next steps, and the time frame.

pchundi, 2022-11-15:
+1 for the implementing Python bindings

pdell-kitware, 2022-11-15:
Hi, 

Ok then !
So I'll start working on it by the end of the week, it should be ready at the latest early next week. 

In the mean-time, to do the binding I used pybind11, the previous version is accessible at: [https://github.com/jedeschaud/ct_icp/tree/old_master/src/binding](https://github.com/jedeschaud/ct_icp/tree/old_master/src/binding)

pchundi, 2022-11-28:
@pdell-kitware: thank you for prioritizing this issue! Anyway I can help?

pdell-kitware, 2022-11-29:
Started working on it, see PR #56 

I'll notify this issue when I am done

pdell-kitware, 2022-12-05:
Hi, So a first version of the binding is completed for `CT-ICP`, 

See branch #56, there are some explanations in the `Readme.md`, 

I'll try to integrate update the binding to pyLiDAR-SLAM in the week (I need to update the binding which have different function names, and methods)

JUN1026, 2023-05-11:
First of all thanks a lot for the author's work, I recently tried ct-icp and pyLiDAR and tried to call ct-icp in pyLiDAR.

I spent a lot of time trying pybind but failed, and finally found out that the project's bindings folder was updated two years ago, so many functions were having problems. In the previous comment, pdell-kitware said that the bindings file will be updated, I would like to know where the last updated file is available? How does the final combination with pyLiDAR work?

I would be grateful if you could answer my questions.

friendship1, 2023-05-31:
> First of all thanks a lot for the author's work, I recently tried ct-icp and pyLiDAR and tried to call ct-icp in pyLiDAR.
>
> I spent a lot of time trying pybind but failed, and finally found out that [...]

same here,
Which is the "latest version" and which is the "first version"?
I couldn't find any branches or tags related to it.