jedeschaud/ct_icp #51: Run ct_icp on Darpa tunnel circuit datasets

Cristian-wp, 2022-11-03:
Hi, I have manage to make the slam work on Darpa urban datasets, now I am trying to test it in the tunnel ciruits.
As you links say, them al compressed, so I decompress all of them with rosbag decompress file.bag .
I have correctly remap the pointcloud topic, lidar model and frequency are the same and at the moment I have not change anything inside the param yaml file.
When I launch the simulation the simulation starts, but the odometry topic /ct_icp/pose/odom is not published.
As you can see from the following images, also the tf from odom to base_link is not generated.

**This image are the TF from the urban datasets:**
[image]

**This one are the TF from the tunnel:**
[image]

**This one is a screenshot from RViz after some seconds:** as you can see the position frame are not published even if the robot is already inside the tunnel (the robot stars outside the tunnel)

[image]

**This one are the topic from rqt:** 
[image]

What I am doing wrong? Can someone please help me?

pdell-kitware, 2022-11-03:
Hi, 

I will look into it,

In the mean time, do you have some error message in the terminal ?

Cristian-wp, 2022-11-03:
Hi, Thank you!
No, in the terminal I have some warning at the launch. But I have read in some post that they are normal.
I use this bag:
https://bitbucket.org/subtchallenge/subt_reference_datasets/src/master/
https://subt-data.s3.amazonaws.com/SubT_Tunnel_Ckt/sr_B_route2.bag (16.3 GB)