jedeschaud/ct_icp #53: Run ct_icp on Jetson Nano

Cristian-wp, 2022-11-14:
Hello, I would like to run this slam on my Jetson Nano 4Gb.
I have manage to install and build on it, but even if I use Ceres as solver, I can not manage to run the solver on the board GPU.
I know that only some type of Ceres option are currently supperted by CUDA:

"[CUDA](https://developer.nvidia.com/cuda-toolkit) If you have an NVIDIA GPU then Ceres Solver can use it accelerate the solution of the Gauss-Newton linear systems using the CMake flag USE_CUDA. Currently this support is limited to using the dense linear solvers that ship with CUDA. As a result GPU acceleration can be used to speed up DENSE_QR, DENSE_NORMAL_CHOLESKY and DENSE_SCHUR. This also enables CUDA mixed precision solves for DENSE_NORMAL_CHOLESKY and DENSE_SCHUR. Optional."

So, I would like to know which dense linear solver have you use.

pdell-kitware, 2022-11-14:
Hello, 

So we use the default Ceres solver (which should be DENSE_QR). 
To be honest, I doubt that this will speedup significantly the process, as the linear system solved if of small size (6 or 12 parameters, thus the matrix to invert is of size up to 12x12). 

What is costly in our method is the construction of the neighborhoods (and the point associations), 
As well as the evaluation of the residuals, rather than solving the Linear Problem.

Cristian-wp, 2022-11-15:
Hello @pdell-kitware thank you for your answer.
In fact I have notice that even on my Deskot PC the slam is a little slow to give the correct output, but in my dataset on a straight path of 250m inside a raillway tunnel it gives 20m of error that is a really good result in this type of test field (concrete flat walls, no signal, no marker,ecc...).

When you run it on a real hardware (in my case is an Ouster OS0-1) do you subsample the pointcloud to speed up?

Another thing, I have see that I can use GTSAM or CERES as solver, based on your experience which is the best in term of solving speed?

Is mandatory to use DENSE_QR as solver or I can try with DENSE_NORMAL_CHOLESKY and DENSE_SCHUR?

Last thing, I am doing these experiments because I wish to run your slam on a drone inside tunnels to make a 3D reconstruction. Have you ever test the slam performance on a real hardware with real sensors(even not on a derone)? For example I am planning to run the slam on the jetsnon and all the other thing on an i7 intel nuc. Do you think is a good choiche?

Cristian-wp, 2022-11-17:
What do you suggest to speed up the lidar odometry?