jedeschaud/ct_icp #55: How GTSAM is used?

Cristian-wp, 2022-11-17:
Hi @jedeschaud ,
as you previously say using CUDA with ceres does not improve the resolution speed in this case.
Now, I have see inside a[https://github.com/jedeschaud/ct_icp/blob/master/CMakeLists.txt](url) that during the compilation GTSAM is searched:

`option(WITH_GTSAM "Search GTSAM dependency (disabled by default)" OFF)`

Even GTSAM can use CUDA with Eigen, so I would like to understand how GTSAM is used in the code.
Thanks in advance.

pdell-kitware, 2022-11-17:
Well... 

**GTSAM** is currently simply not used as of this time in the code. 

We are preparing a Loop Closure release with a Factor-Graph based optimization built using GTSAM. 
But for now, the repo only has a LiDAR Odometry, which uses Ceres for optimization in the ICP. 

Hope this clarifies your questions.

Cristian-wp, 2022-11-17:
Perfectly clear!

closed 2022-11-17 by pdell-kitware