jedeschaud/ct_icp #57: Modify superbuild ceres version

Cristian-wp, 2022-12-12:
Hello,
is possible to modify the Ceres version downloaded during the dependency installation from `https://gitlab.kitware.com/keu-computervision/MappingResearchKEU/Superbuild ` ?

At the moment Kitware use Ceres 2.0.0, but I wish to use the latest stable version (2.1.0).
How can I do?

pierdell, 2022-12-12:
Hi, 

So yes it is possible, 

The best way to do this is to clone the repository 

https://gitlab.kitware.com/keu-computervision/MappingResearchKEU/Superbuild

And before building, modify the CMakeLists.txt file to specify the Ceres version:

See the handling of Ceres depency:
https://gitlab.kitware.com/keu-computervision/MappingResearchKEU/Superbuild/-/blob/master/CMakeLists.txt#L128

You need to replace the GIT_TAG 2.0.0 by GIT_TAG 2.1.0

And then you can follow the instructions of the project to build the Superbuild. 

/!\ The superbuild will install many libraries you absolutely don't need !

So make sure to pass the same options as CT_ICP's superbuild/CMakeLists.txt 

ie:

        -DDEFAULT_LIB_OPTION=OFF
        -DWITH_GLOG=ON
        -DWITH_EIGEN=ON
        -DWITH_CERES=ON
        -DWITH_YAML_CPP=ON
        -DWITH_GTEST=ON
        -DWITH_TCLAP=ON
        -DWITH_TESSIL=ON
        -DWITH_TINYPLY=ON
        -DWITH_NANOFLANN=ON
        -DWITH_COLORMAP=ON
        -DWITH_CEREAL=ON

Cristian-wp, 2022-12-13:
Hey @pierdell thank you for your answer. I was able to do it before your answer, but the steps were exactly the same you suggest!
Thanks a lot :)