jedeschaud/ct_icp #64: An error occurred while compiling:  /usr/bin/ld

zyqww, 2023-02-16:
Hi authors,thanks for the great work, but I got a new error at step 2 when building the ct-icp library.The following is what I have done:

      $ cmake --build . --target install --config Release --parallel 12

Then the prompt is

       /usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_yaml_utils.dir/build.make:107：test/unit/SlamCore/test_yaml_utils] 错误 1
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_pointcloud.dir/build.make:107：test/unit/SlamCore/test_pointcloud] 错误 1
make[1]: *** [CMakeFiles/Makefile2:438：test/unit/SlamCore/CMakeFiles/test_pointcloud.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/Makefile2:542：test/unit/SlamCore/CMakeFiles/test_yaml_utils.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_geometry.dir/build.make:107：test/unit/SlamCore/test_geometry] 错误 1
make[1]: *** [CMakeFiles/Makefile2:464：test/unit/SlamCore/CMakeFiles/test_geometry.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_imu.dir/build.make:107：test/unit/SlamCore/test_imu] 错误 1
make[1]: *** [CMakeFiles/Makefile2:386：test/unit/SlamCore/CMakeFiles/test_imu.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_A_grid_sampling.dir/build.make:107：test/unit/SlamCore/test_A_grid_sampling] 错误 1
make[1]: *** [CMakeFiles/Makefile2:412：test/unit/SlamCore/CMakeFiles/test_A_grid_sampling.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_trajectory.dir/build.make:107：test/unit/SlamCore/test_trajectory] 错误 1
make[1]: *** [CMakeFiles/Makefile2:672：test/unit/SlamCore/CMakeFiles/test_trajectory.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_eval.dir/build.make:107：test/unit/SlamCore/test_eval] 错误 1
make[1]: *** [CMakeFiles/Makefile2:620：test/unit/SlamCore/CMakeFiles/test_eval.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_config.dir/build.make:107：test/unit/SlamCore/test_config] 错误 1
make[1]: *** [CMakeFiles/Makefile2:568：test/unit/SlamCore/CMakeFiles/test_config.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_neighborhood.dir/build.make:107：test/unit/SlamCore/test_neighborhood] 错误 1
make[1]: *** [CMakeFiles/Makefile2:490：test/unit/SlamCore/CMakeFiles/test_neighborhood.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_io.dir/build.make:107：test/unit/SlamCore/test_io] 错误 1
make[1]: *** [CMakeFiles/Makefile2:516：test/unit/SlamCore/CMakeFiles/test_io.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_timer.dir/build.make:107：test/unit/SlamCore/test_timer] 错误 1
make[1]: *** [CMakeFiles/Makefile2:698：test/unit/SlamCore/CMakeFiles/test_timer.dir/all] 错误 2
/usr/bin/ld: warning: libglog.so.0.3.5, needed by ../../../../install/Ceres/lib/libceres.so.2.0.0, not found (try using -rpath or -rpath-link)
/usr/bin/ld: ../../../../install/Ceres/lib/libceres.so.2.0.0: undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [test/unit/SlamCore/CMakeFiles/test_buffer.dir/build.make:107：test/unit/SlamCore/test_buffer] 错误 1
make[1]: *** [CMakeFiles/Makefile2:724：test/unit/SlamCore/CMakeFiles/test_buffer.dir/all] 错误 2
make: *** [Makefile:136：all] 错误 2

How should I solve it?  Thank you!

pierdell, 2023-02-19:
Hi, 

For some reason, ceres does not find glog. 
However the superbuild should have installed it along all other external dependencies. 
Can you check in the directory */install if glog has been installed ?

qsisi, 2023-03-07:
> Hi,
> 
> For some reason, ceres does not find glog. However the superbuild should have installed it along all other external dependencies. Can you check in the directory */install if glog has been installed ?
> 
> This type of problem typically occur when a "system" version of glog is installed which is selected instead of the glog built by the superbuild.

I meet the same error, the glog is not installed under the directory */install, I guess it's because I have already installed glog before when dealing with other projects, so the superbuild doesn't download the glog.