jedeschaud/ct_icp #66: Fail to copy libglog during superbuild

gitouni, 2023-03-25:
Thank you for sharing your advanced work. I met an issue during step1. It seems that the installed `glog` in my system conflicts with the `glog` linked by superbuild.
# Environment
| System | ROS| GCC | Cmake |
|----|-----|----| --- |
|Ubuntu 20.04 | noetic | 9.4.0 | 3.25.0 |
# Issue
Issue appeared when using command `cmake --build . --config Release ` in Step 1
Output (part)
```
[ 36%] Linking CXX shared library libglog.so
/usr/bin/ld: /usr/local/lib/libgflags.a(gflags.cc.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: 最后的链结失败: bad value
collect2: error: ld returned 1 exit status
make[8]: *** [CMakeFiles/glog.dir/build.make:194：libglog.so.0.5.0] 错误 1
make[7]: *** [CMakeFiles/Makefile2:853：CMakeFiles/glog.dir/all] 错误 2
make[6]: *** [Makefile:166：all] 错误 2
make[5]: *** [CMakeFiles/glog.dir/build.make:86：glog/src/glog-stamp/glog-build] 错误 2
make[4]: *** [CMakeFiles/Makefile2:154：CMakeFiles/glog.dir/all] 错误 2
make[3]: *** [Makefile:136：all] 错误 2
make[2]: *** [CMakeFiles/MappingResearchKEU_superbuild.dir/build.make:86：MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-build] 错误 2
make[1]: *** [CMakeFiles/Makefile2:83：CMakeFiles/MappingResearchKEU_superbuild.dir/all] 错误 2
make: *** [Makefile:91：all] 错误 2

```
# Possible Cause
As the bug report, relocation `R_X86_64_PC32` against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC

fratopa, 2023-03-28:
Moving from Gcc 9.4.0 to gcc 11.1.0 solved the glog problem, I still have more but its a step forward I guess. You can check [issue 67 ](https://github.com/jedeschaud/ct_icp/issues/67) to check where I get stuck now

gitouni, 2023-03-29:
I appreciate the replies from @pierdell  and @fratopa. I have solved this problem in a manual way.

First, add `-DWITH_GFLAGS=OFF` config in [.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild/CMakeLists.txt](https://gitlab.kitware.com/keu-computervision/MappingResearchKEU/Superbuild/-/blob/master/CMakeLists.txt#L85) does fix the bug in the Step 1, but  the same bug emegered at my step 2. To fix it, we need to manually change the static link files to the dynamic ones in `link.txt`.

## 1. First, run the first two commands of Step 2:
```bash
mkdir cmake-build-release && cd  cmake-build-release  
cmake .. -DCMAKE_BUILD_TYPE=Release 
```
## 2. Second, manually modify the `link.txt` files.
* For `SlamCore`:
```bash
cd /path/to/root/
cd cmake-build-release/src/ct_icp/CMakeFiles/CT_ICP.dir
gedit link.txt
```
Then, substitute each `libglog.a` with `libglog.so`.
* The similar operation for `ct_icp`:
```bash
cd /path/to/root/
cd cmake-build-release/src/ct_icp/CMakeFiles/CT_ICP.dir
gedit link.txt
```
Then, substitute each `libgflas.a` with `libgflags.so`.
## 3. run the third command line of Step 2
```
cmake --build . --target install --config Release -j 12 
```

## Cmake Passed!
```log
...
-- Up-to-date: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/include/SlamCore/cereal.h
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/target_utils.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/utils.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/test_utils.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/includes.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/project_config.cmake.in
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/install_utils.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/viz3d_utils.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/superbuild_utils.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/external.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib/cmake/cmake/flags.cmake
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/test/all_tests
-- Set runtime path of "/home/bit/CODE/Drivers/ct_icp/install/CT_ICP/test/all_tests" to ":/usr/local/lib:/usr/local/lib:/home/bit/CODE/Drivers/ct_icp/install/yaml-cpp/lib:/home/bit/CODE/Drivers/ct_icp/install/tinyply/lib:/home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib:/home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib:/usr/local/lib:/usr/local/cuda-11.1/lib64"
-- Installing: /home/bit/CODE/Drivers/ct_icp/install/CT_ICP/bin/run_odometry
-- Set runtime path of "/home/bit/CODE/Drivers/ct_icp/install/CT_ICP/bin/run_odometry" to ":/usr/local/lib:/usr/local/lib:/home/bit/CODE/Drivers/ct_icp/install/yaml-cpp/lib:/home/bit/CODE/Drivers/ct_icp/install/tinyply/lib:/home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib:/home/bit/CODE/Drivers/ct_icp/install/CT_ICP/lib:/usr/local/cuda-11.1/lib64:/usr/local/lib"
```

closed 2023-03-29 by gitouni