jedeschaud/ct_icp #67: Superbuild fail

fratopa, 2023-03-28:
Hi, I am having a lot of problems compiling this repo. I am running ubuntu 20.04, ROS Noetic, Gcc 11.1.0 and cmake 3.26.1. 
I issue the following commands:
```
git clone https://github.com/jedeschaud/ct_icp
cd ct_icp
mkdir .cmake-build-superbuild && cd .cmake-build-superbuild
cmake ../superbuild
cmake --build . --config Release
```
everything seems to be going well except for some minor warnings such as 
```
CMake Warning (dev) at /home/MyUserName/.local/lib/python3.8/site-packages/cmake/data/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (SUPERLU)
  does not match the name of the calling package (SuperLU).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindSuperLU.cmake:93 (find_package_handle_standard_args)
  bench/spbench/CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
```
However, when trying to build ceres everything crashes down:
```
[ 95%] Building CXX object internal/ceres/CMakeFiles/small_blas_gemm_benchmark.dir/small_blas_gemm_benchmark.cc.o
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/small_blas_gemv_benchmark.cc:81:6: warning: no previous declaration for ‘void ceres::BM_MatrixVectorMultiply(benchmark::State&)’ [-Wmissing-declarations]
   81 | void BM_MatrixVectorMultiply(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/small_blas_gemv_benchmark.cc:97:6: warning: no previous declaration for ‘void ceres::BM_MatrixTransposeVectorMultiply(benchmark::State&)’ [-Wmissing-declarations]
   97 | void BM_MatrixTransposeVectorMultiply(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/schur_eliminator_benchmark.cc:130:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorEliminate(benchmark::State&)’ [-Wmissing-declarations]
  130 | void BM_SchurEliminatorEliminate(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/schur_eliminator_benchmark.cc:153:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorBackSubstitute(benchmark::State&)’ [-Wmissing-declarations]
  153 | void BM_SchurEliminatorBackSubstitute(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/schur_eliminator_benchmark.cc:181:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorForOneFBlockEliminate(benchmark::State&)’ [-Wmissing-declarations]
  181 | void BM_SchurEliminatorForOneFBlockEliminate(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/schur_eliminator_benchmark.cc:195:6: warning: no previous declaration for ‘void ceres::internal::BM_SchurEliminatorForOneFBlockBackSubstitute(benchmark::State&)’ [-Wmissing-declarations]
  195 | void BM_SchurEliminatorForOneFBlockBackSubstitute(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 96%] Linking CXX executable ../../bin/small_blas_gemv_benchmark
[ 97%] Linking CXX executable ../../../bin/autodiff_benchmarks
[ 98%] Linking CXX executable ../../bin/schur_eliminator_benchmark
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/invert_psd_matrix_benchmark.cc:65:6: warning: no previous declaration for ‘void ceres::internal::BenchmarkDynamicallyInvertPSDMatrix(benchmark::State&)’ [-Wmissing-declarations]
   65 | void BenchmarkDynamicallyInvertPSDMatrix(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/small_blas_gemm_benchmark.cc:89:6: warning: no previous declaration for ‘void ceres::internal::BM_MatrixMatrixMultiplyDynamic(benchmark::State&)’ [-Wmissing-declarations]
   89 | void BM_MatrixMatrixMultiplyDynamic(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/MyUserName/ct_icp_NEW/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/internal/ceres/small_blas_gemm_benchmark.cc:135:6: warning: no previous declaration for ‘void ceres::internal::BM_MatrixTransposeMatrixMultiplyDynamic(benchmark::State&)’ [-Wmissing-declarations]
  135 | void BM_MatrixTransposeMatrixMultiplyDynamic(benchmark::State& state) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 99%] Linking CXX executable ../../../bin/autodiff_benchmarks_fast_math
[100%] Linking CXX executable ../../bin/invert_psd_matrix_benchmark
[100%] Linking CXX executable ../../bin/small_blas_gemm_benchmark
/usr/bin/ld: CMakeFiles/schur_eliminator_benchmark.dir/schur_eliminator_benchmark.cc.o: in function `ceres::internal::BM_SchurEliminatorEliminate(benchmark::State&)':
schur_eliminator_benchmark.cc:(.text+0x6c): undefined reference to `ceres::internal::ThreadPool::ThreadPool()'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0x10f): undefined reference to `ceres::internal::SchurEliminatorBase::Create(ceres::internal::LinearSolver::Options const&)'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0x12b): undefined reference to `ceres::internal::BlockSparseMatrix::block_structure() const'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0x39e): undefined reference to `ceres::internal::ThreadPool::~ThreadPool()'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0x47d): undefined reference to `ceres::internal::BlockSparseMatrix::block_structure() const'
/usr/bin/ld: CMakeFiles/schur_eliminator_benchmark.dir/schur_eliminator_benchmark.cc.o: in function `ceres::internal::BM_SchurEliminatorBackSubstitute(benchmark::State&)':
schur_eliminator_benchmark.cc:(.text+0x69c): undefined reference to `ceres::internal::ThreadPool::ThreadPool()'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0x73f): undefined reference to `ceres::internal::SchurEliminatorBase::Create(ceres::internal::LinearSolver::Options const&)'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0x756): undefined reference to `ceres::internal::BlockSparseMatrix::block_structure() const'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0x956): undefined reference to `ceres::internal::BlockSparseMatrix::block_structure() const'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0xa30): undefined reference to `ceres::internal::ThreadPool::~ThreadPool()'
/usr/bin/ld: schur_eliminator_benchmark.cc:(.text+0xaec): undefined reference to `ceres::internal::BlockSparseMatrix::block_structure() const'
[... 17 lines cut ...]
schur_eliminator_benchmark.cc:(.text.unlikely+0x92): undefined reference to `ceres::internal::ThreadPool::~ThreadPool()'
collect2: error: ld returned 1 exit status
make[8]: *** [internal/ceres/CMakeFiles/schur_eliminator_benchmark.dir/build.make:102: bin/schur_eliminator_benchmark] Error 1
[100%] Built target small_blas_gemm_benchmark
make[7]: *** [CMakeFiles/Makefile2:229: internal/ceres/CMakeFiles/schur_eliminator_benchmark.dir/all] Error 2
make[7]: *** Waiting for unfinished jobs....
[100%] Built target small_blas_gemv_benchmark
[100%] Built target invert_psd_matrix_benchmark
[100%] Built target autodiff_benchmarks_fast_math
[100%] Built target autodiff_benchmarks
make[6]: *** [Makefile:146: all] Error 2
make[5]: *** [CMakeFiles/Ceres.dir/build.make:86: Ceres/src/Ceres-stamp/Ceres-build] Error 2
make[4]: *** [CMakeFiles/Makefile2:260: CMakeFiles/Ceres.dir/all] Error 2
[... cut ...]
make[2]: *** [CMakeFiles/MappingResearchKEU_superbuild.dir/build.make:86: MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-stamp/MappingResearchKEU_superbuild-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/MappingResearchKEU_superbuild.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```
I have tried with clang 10 and got the same error. I also tried most of the alternative branches of this repo 
- master
- python_bindings
- pdell-kitware-patch-1
- fix_#37
and none of them solved my problem. Do you have any suggestions on what to do? 
Thanks in advance to anyone in my situation

fratopa, 2023-03-28:
I believe it may be useful to show how cmake was configuring everything to compile ceres:
```
[ 48%] Performing configure step for 'Ceres'
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detected Ceres being used as a git submodule, adding commit hook for Gerrit to: /home/MyUserName/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/.git
-- Detected Ceres version: 2.0.0 from /home/MyUserName/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres/include/ceres/version.h
-- Detected available Ceres threading models: [CXX_THREADS, OPENMP, NO_THREADS]
-- Building with C++17
-- Found Eigen version 3.3.7: /home/MyUserName/ct_icp/install/Eigen3/include/eigen3
-- Enabling use of Eigen as a sparse linear algebra library.
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found LAPACK library: /usr/lib/x86_64-linux-gnu/liblapack.so;/usr/lib/x86_64-linux-gnu/libblas.so;/usr/lib/x86_64-linux-gnu/libf77blas.so;/usr/lib/x86_64-linux-gnu/libatlas.so
-- Found AMD headers in: /usr/include/suitesparse
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found CAMD headers in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found COLAMD headers in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found CCOLAMD headers in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CHOLMOD headers in: /usr/include/suitesparse
-- Found CHOLMOD library: /usr/lib/x86_64-linux-gnu/libcholmod.so
-- Found SUITESPARSEQR headers in: /usr/include/suitesparse
-- Found SUITESPARSEQR library: /usr/lib/x86_64-linux-gnu/libspqr.so
-- Found Intel Thread Building Blocks (TBB) library (2020.1 / 11101) include location: . Assuming SuiteSparseQR was compiled with TBB.
-- Found SUITESPARSE_CONFIG headers in: /usr/include/suitesparse
-- Found SUITESPARSE_CONFIG library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found LIBRT library: /usr/lib/x86_64-linux-gnu/librt.so
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: TRUE (found version "5.7.1") 
-- Found SuiteSparse 5.7.1, building with SuiteSparse.
-- Found CXSparse: /usr/include/suitesparse (found version "3.2.0") 
-- Found CXSparse version: 3.2.0, building with CXSparse.
-- Building without Apple's Accelerate sparse support.
-- Found Google Flags (gflags) version 2.2.2: /usr/lib/x86_64-linux-gnu/cmake/gflags
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Found exported glog build directory: /home/MyUserName/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/glog/src/glog-build
-- Detected glog version: 0.5.0
-- Found Glog: glog::glog  
-- Found Google Log (glog). Assuming glog was built with gflags support as gflags was found. This will make gflags a public dependency of Ceres.
-- Using Ceres threading model: CXX_THREADS
-- Found Google benchmark library. Building Ceres benchmarks.
-- Building Ceres as a shared library.
-- Performing Test CHECK_CXX_FLAG_Wmissing_declarations
-- Performing Test CHECK_CXX_FLAG_Wmissing_declarations - Success
-- Performing Test CHECK_CXX_FLAG_Wno_unknown_pragmas
-- Performing Test CHECK_CXX_FLAG_Wno_unknown_pragmas - Success
-- Performing Test CHECK_CXX_FLAG_Wno_sign_compare
-- Performing Test CHECK_CXX_FLAG_Wno_sign_compare - Success
-- Performing Test CHECK_CXX_FLAG_Wno_unused_parameter
-- Performing Test CHECK_CXX_FLAG_Wno_unused_parameter - Success
-- Performing Test CHECK_CXX_FLAG_Wno_missing_field_initializers
-- Performing Test CHECK_CXX_FLAG_Wno_missing_field_initializers - Success
-- Creating configured Ceres config.h output directory: /home/MyUserName/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres-build/config/ceres/internal
-- Enabling CERES_USE_EIGEN_SPARSE in Ceres config.h
-- Enabling CERES_NO_ACCELERATE_SPARSE in Ceres config.h
-- Enabling CERES_USE_CXX_THREADS in Ceres config.h
-- Enabling CERES_USING_SHARED_LIBRARY in Ceres config.h
-- Performing Test CHECK_CXX_FLAG_Wno_missing_declarations
-- Performing Test CHECK_CXX_FLAG_Wno_missing_declarations - Success
-- Do not build any example.
-- Configuring done (3.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/MyUserName/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild-build/Ceres/src/Ceres-build
```
Maybe someone can pinpoint the bug

pierdell, 2023-03-28:
Hi, 

So you seem not to be able to build and install ceres, which is weird, never had this problem before. 

My advice would be to : 

1. first try and install ceres independently on your machine through cmake (see the ceres github repo)
2. If it succeeds, that is weird, but quite indicative and you can probably use the ceres version instead of the superbuild (I'll guide you through the cmake if need be)
3. If it fails, try a different ceres branch ?

Let me know, 

Cheers

fratopa, 2023-03-28:
Hi! In the end, I managed to solve the problem by changing the ceres version from 2.0.0 to 2.1.0 in the CMakeList.txt file `~/ct_icp/.cmake-build-superbuild/MappingResearchKEU_superbuild/src/MappingResearchKEU_superbuild\CMakeLists.txt` Line 137.

My best guess is that having already installed ceres 2.1.0 previously messed something up in the linker part of the compilation. Either that or there are some incompatibilities between the compiler itself and ceres 2.0.0 which I find not that likely. 

Before closing the issue I will try also to build the ROS portion of the repo. In the meantime, I kindly ask you to leave the issue open.

fratopa, 2023-03-29:
OK, I was finally able to build the project with ROS. the key takeaway points are that it's best to use the same ceres version you may have previously used and always remember to source the ros workspace before compiling.

closed 2023-03-29 by fratopa