jedeschaud/ct_icp #68: ct_icp no timestamps

fratopa, 2023-03-29:
Hi! I am trying to run your algorithm on some data I generate from a gazebo simulation. Such data do not come with timestamps associated with the points and I would like to know if the algorithm can still run on them. 
At the moment when I try and launch the file I get the error: 
```
The point cloud does not have timestamps, this is incompatible with the `CONTINUOUS_TIME` representation of pose in CT-ICP
[ INFO] [1680086675.307007042]: The schema of the point cloud is:
Item Size: 32
Elements Layout:
	properties: [offset in item:0]
		x: [type:f / offset in elem:0 / dim:1]
		y: [type:f / offset in elem:4 / dim:1]
		z: [type:f / offset in elem:8 / dim:1]
		padding_0: [type:b / offset in elem:12 / dim:4]
		intensity: [type:f / offset in elem:16 / dim:1]
		tag: [type:B / offset in elem:20 / dim:1]
		line: [type:B / offset in elem:21 / dim:1]
		padding_1: [type:b / offset in elem:22 / dim:10]
	vertex: [offset in item:0]
		x: [type:f / offset in elem:0 / dim:1]
		y: [type:f / offset in elem:4 / dim:1]
		z: [type:f / offset in elem:8 / dim:1]
```
What parameter should I change from CONTINUOUS_TIME to something else?

pierdell, 2023-03-30:
You should change it to "CONSTANT_VELOCITY" / "ITERATIVE", 

Depending on your use case, you should try either 
 - the driving config:
[config/driving_config.yaml](https://github.com/jedeschaud/ct_icp/blob/master/config/odometry/driving_config.yaml)
Or the mobile robotics config:
[config/driving_config.yaml](https://github.com/jedeschaud/ct_icp/blob/master/config/odometry/driving_config.yaml)
 - https://github.com/jedeschaud/ct_icp/blob/master/config/odometry/nclt_config.yaml

fratopa, 2023-03-30:
Hi, thank you for pointing me in the right direction. I am using CT-ICP on ROS and if I try and set the parametrization value to `CONSTANT_VELOCITY" / "ITERATIVE` I get the error: 
```
Loading Config from yaml file: /home/MyUserName/ct_icp_ros/src/ct_icp_odometry/params/ct_icp/ct_icp_driving_fast.yaml
[ERROR] [1680179983.763600138]: Error while loading the config from path: `/home/MyUserName/ct_icp_ros/src/ct_icp_odometry/params/ct_icp/ct_icp_driving_fast.yaml`
inotify_add_watch(/home/MyUserName/.config/ibus/bus/5f473cec13374b419a0b78498fa8d504-unix-1) failed: (No space left on device)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Parametrization is not supported !
Caught SIGABRT: usually caused by an abort() or assert()
  message for depth: [2]: /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7fa81862d00b]
  message for depth: [3]: /lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7fa81860c859]
  message for depth: [4]: /lib/x86_64-linux-gnu/libstdc++.so.6(+0xa1a31) [0x7fa8189e9a31]
  message for depth: [5]: /lib/x86_64-linux-gnu/libstdc++.so.6(+0xad5dc) [0x7fa8189f55dc]
  message for depth: [6]: /lib/x86_64-linux-gnu/libstdc++.so.6(+0xad647) [0x7fa8189f5647]
  message for depth: [7]: /lib/x86_64-linux-gnu/libstdc++.so.6(__cxa_rethrow+0x4d) [0x7fa8189f593d]
  message for depth: [8]: /home/MyUserName/ct_icp_ros/devel/lib/ct_icp_odometry/ct_icp_odometry_node(+0x153b6) [0x55e6477073b6]
  message for depth: [9]: /home/MyUserName/ct_icp_ros/devel/lib/ct_icp_odometry/ct_icp_odometry_node(+0x16aa3) [0x55e647708aa3]
  message for depth: [10]: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fa81860e083]
[ct_icp_odometry_node-3] process has died [pid 184369, exit code 1, cmd /home/MyUserName/ct_icp_ros/devel/lib/ct_icp_odometry/ct_icp_odometry_node /ct_icp/pointcloud:=/livox/lidar __name:=ct_icp_odometry_node __log:=/home/MyUserName/.ros/log/f200a7be-cef7-11ed-aef6-bb7807375447/ct_icp_odometry_node-3.log].
log file: /home/MyUserName/.ros/log/f200a7be-cef7-11ed-aef6-bb7807375447/ct_icp_odometry_node-3*.log
```
However I can set `parametrization: SIMPLE` and it works very well, you can see below the results using my data where the pointclouds have no timestamp data: 
[image]

pierdell, 2023-03-30:
Nice ! Thanks for sharing :)