jedeschaud/ct_icp #69: Running CT-ICP with other Datasets on ROS

cigdemkknz, 2023-04-03:
Hello authors, great work!

I was hoping to run it on this KITTI city dataset (https://www.cvlibs.net/datasets/kitti/raw_data.php  -urban driving) and my own dataset collected with Velodyne 32C (outdoor mobile robot) using ROS. I have made the attached config files for running these two datasets. The KITTI one is based on "driving_config.yaml" and mine is based on "nclt_config.yaml". Since there are no timestamps on these data, and CONSTANT_VELOCITY and ITERATIVE values gives errors, I use the "SIMPLE" parametrization. 

Both datasets quits after a short while with the following terminal error.

---------------------------------------------------------------
QObject::connect: Cannot queue arguments of type 'QVector<int>'7.858233               6497.65 
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
Error in ego-motion distance !65.885353   Duration: 1.567941 / 7.858233               
[ct_icp_odometry_node-3] process has finished cleanly.762894 / 7.858233               
---------------------------------------------------------------

Any guidance on how to solve these issues and what parameters I should use would be greatly appreciated!

Great work!

[yaml_files.zip](https://github.com/jedeschaud/ct_icp/files/11143170/yaml_files.zip)
[image]
[image]

Tianci-Wen, 2023-05-29:
same problem! Have you solved it yet? Thanks!

tmddn833, 2023-12-13:
Change the motion_compensation to ITERATIVE in config file. It works for me.
  motion_compensation: ITERATIVE # The profile of the motion compensation (NONE, CONSTANT_VELOCITY, ITERATIVE, CONTINUOUS)