jedeschaud/ct_icp #70: Trajectory Location and Replicating Kitti Raw Results

cigdemkknz, 2023-04-04:
Hello authors,

I am trying to replicate the Kitti raw results using the scripts (OPTION 1). I am struggling to locate the saved trajectory, as the trajectory PLY path on the terminal doesn't exist after SLAM completes. For reference here are the terminal output and the visualization. Also it shows SLAM has failed and unfinished. 

Any hints on how to address these would be greatly appreciated!

[image]
[image]

pierdell, 2023-04-13:
Hi, 

So do you mean that the trajectory is not saved at the location indicated by the console ?

cigdemkknz, 2023-04-13:
Hello, 

1. Thank you for the response. We were able to save the trajectory once we changed the output directory from the config file! We are able to evaluate the results. Using the evo package we were able to generate an ATE of 4.29m on the KITTI_raw sequence 00, using the driving_config.yaml with GN solver, and the GN solver Specific parameters. The remaining parameters are used same as in the master branch. The result is better than the reported result of 6.22m in the paper. 

-Are we running in the same setup as your paper's KITTI raw parameters?
-Has the parameters been fine tuned since the release of the paper?

2. We were hoping to run the nclt dataset as well, similarly to how we are running the KITTI raw, using the nclt_config.yaml. However this time we are getting this error:

"Check failed: sequence_names.size() == sequence_infos.size() Could not find all the sequence in "sequence_options"

The config file we are using is attached. Should we proceed with addressing the errors, or is there a better way to run the nclt datasets?

3. Has the point-to-point and point-to-distribution metrics been implemented, or is it currently all using point-to-plane? Changing these parameters didn't seem to have an impact on the KITTI raw 00 sequence trajectory results.

4. Finally, what dataset would you recommend for testing the motion constraints (Cloc and Cvel). Removing the motion constraints didn't alter the trajectory results for the KITTI raw 00 for us. Perhaps not enough motion changes with this dataset?

These are a lot of questions, please feel free to answer them at your convenience,

Excellent work!! Results are amazing even without loop closure. 

[nclt_config.zip](https://github.com/jedeschaud/ct_icp/files/11226641/nclt_config.zip)