jedeschaud/ct_icp #71: How to obtain the relative timestamps?

Hero941215, 2023-05-17:
Hello authors,

You have completed a great job!

During the process of studying your article and code, I found that the relative timestamp range of each point you provided is [0,1]. The classic LOAM algorithm also seems to have a method for calculating relative timestamps (they use the first and last points of a frame  to determine the angle at the beginning and end of the frame), and the calculated timestamps are not within the range of [0,1].

Therefore, I would like to know how you process the original point cloud data to obtain the relative timestamp of each point within the range of [0,1]?

Hero941215, 2023-05-21:
> Thank you very much for your reply.

According to the formula you provided, when we can obtain the real timestamp of each point in a frame of point cloud, the relative timestamp can be further calculated, and the range is within [0,1].

However, the point cloud in KITTI-raw is stored in txt format and provides [x, y, z, intensity]. So, how is the real timestamp ta obtained?

pierdell, 2023-05-26:
We provide a link to the raw-kitti datasets (which have timestamps). 

To obtain this timestamps, we took the azimuth (yaw) angle and linearly interpolated between 0 and 1 
Starting 0 towards the back of the car (180° with respect to the X-Axis), and going clockwise.

See the picture of the timestamps (blue means 0, red 1):

 
[image]

Hero941215, 2023-05-26:
> We provide a link to the raw-kitti datasets (which have timestamps).
>
> To obtain this timestamps, we took the azimuth (yaw) angle and linearly interpolated between 0 and 1 Starting 0 towards the back [...]

Thank you very much. I understand how the timestamp was obtained