jedeschaud/ct_icp #74: Error during superbuild installation

EdoardoPassera, 2023-07-07:
I get the following error during superbuild installation (step 1):
Found Unwind: /usr/include/x86_64-linux-gnu (found suitable version "1.6.2", minimum required is "1.3.2") 
CMake Error at CMakeLists.txt:432 (message):
Can't find Google Log (glog).  Please set either: glog_DIR (newer CMake
built versions of glog) or GLOG_INCLUDE_DIR & GLOG_LIBRARY or enable
MINIGLOG option to use minimal glog implementation.

fghoussen, 2023-08-19:
Same problem. Setting `export CMAKE_MODULE_PATH="/usr/lib/x86_64-linux-gnu/cmake/glog:${CMAKE_MODULE_PATH}"` doesn't help. Any workaround?

aidan-ferguson, 2023-11-10:
For me this was an issue with having the glog development library already installed as I believe the superbuild script attempts to build glog from source:

`apt remove libgoogle-glog-dev` & `apt autoremove` did the trick for me

themightyoarfish, 2024-09-27:
its odd that the `-DMINILOG` and `-Dglog_DIR` cmake options have no effect, contrary to the error message.

ignotuspeverel, 2025-03-19:
> Actually it seems to run for me now too, after removing `libgoogle-glog*`.

I remove libgoogle-glog but still got the problem, any idea of fixing this problem ?