jedeschaud/ct_icp #75: INTERFACE_LIBRARY targets may only have whitelisted properties.  The   property "IMPORTED_RELEASE_LOCATION" is not allowed

slamjingjing, 2023-08-09:


KeyNG666, 2023-08-16:
update cmake

asackan, 2024-02-20:
I changed  cmake/external.cmake 13 line. so I fixed it
```cmake
get_target_property(__LOCATION ${target} IMPORTED_RELEASE_LOCATION)
```
to
```cmake
get_target_property(__LOCATION ${target} IMPORTED_LOCATION_RELEASE)
```

saadehmd, 2024-07-10:
INFO /ros_ws/src/ct_icp/install/Ceres/lib/libceres.so.2.0.0
CMake Error at cmake/external.cmake:13 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_RELEASE" is not allowed.
Call Stack (most recent call first):
  CMakeLists.txt:51 (include)

INFO /ros_ws/src/ct_icp/install/yaml-cpp/lib/libyaml-cpp.so.0.6.3
CMake Error at cmake/external.cmake:13 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_RELEASE" is not allowed.
Call Stack (most recent call first):
  CMakeLists.txt:51 (include)

saadehmd, 2024-07-10:
Updating cmake to version 3.20.5 solves this issue :)