jedeschaud/ct_icp #82: Running ct_icp on KITTI

laurenzheidrich, 2024-05-29:
Hello!

I am trying to set-up ct_icp and give it an initial run on the KITTI odometry dataset. I have downloaded it from the official kitti source and converted the .bin velodyne files to .ply and saved them in the velodyne folder. I renamed the original velodyne folder containing the .bin files

Now I want to run your method by:

./install/CT_ICP/bin/run_odometry -c config/odometry/driving_config.yaml

In the config I have changed the dataset option to:

dataset_options:
  - dataset: KITTI
    root_path: /mnt/hdd/KITTI_ODOMETRY/sequences/
    use_all_datasets: false
    sequence_options:
      - sequence_name: 07

If at all possible I would not download your KITTI datset as this would occupy another 70GB of space, when I already have KITTI downloaded.

When running this, I always get the following error: 

I20240529 12:15:46.457690 3322870 odometry_runner.cpp:99] [File: odometry_runner.cpp, Line: 99] -- Running the odometry on all the following sequences from the Dataset:
I20240529 12:15:46.457703 3322870 odometry_runner.cpp:102] [File: odometry_runner.cpp, Line: 102] -- Name: 07, Num Frames=1100 - With  Ground Truth Poses
I20240529 12:15:46.457782 3322870 odometry_runner.cpp:129] Running the Odometry on the sequence named KITTI_07
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append
Caught SIGABRT: usually caused by an abort() or assert()

This really isn't very informative to me right now, so I don't really know how to proceed. I will paste the complete terminal output in the comments. I would greatly appreciate some help! Thanks!

laurenzheidrich, 2024-05-29:
WARNING: Logging before InitGoogleLogging() is written to STDERR
I20240529 12:15:46.365061 3322870 dataset.cpp:1092] Could not load ground truth for sequence 18. Expected at location 
I20240529 12:15:46.375315 3322870 dataset.cpp:1092] Could not load ground truth for sequence 11. Expected at location 
I20240529 12:15:46.429157 3322870 dataset.cpp:1092] Could not load ground truth for sequence 14. Expected at location 
I20240529 12:15:46.429169 3322870 dataset.cpp:1092] Could not load ground truth for sequence 17. Expected at location 
I20240529 12:15:46.429175 3322870 dataset.cpp:1092] Could not load ground truth for sequence 13. Expected at location 
I20240529 12:15:46.441386 3322870 dataset.cpp:1092] Could not load ground truth for sequence 16. Expected at location 
I20240529 12:15:46.441396 3322870 dataset.cpp:1092] Could not load ground truth for sequence 20. Expected at location 
I20240529 12:15:46.441403 3322870 dataset.cpp:1092] Could not load ground truth for sequence 12. Expected at location 
I20240529 12:15:46.441408 3322870 dataset.cpp:1092] Could not load ground truth for sequence 21. Expected at location 
I20240529 12:15:46.441414 3322870 dataset.cpp:1092] Could not load ground truth for sequence 15. Expected at location 
I20240529 12:15:46.441419 3322870 dataset.cpp:1092] Could not load ground truth for sequence 19. Expected at location 
I20240529 12:15:46.457690 3322870 odometry_runner.cpp:99] [File: odometry_runner.cpp, Line: 99] -- Running the odometry on all the following sequences from the Dataset:
I20240529 12:15:46.457703 3322870 odometry_runner.cpp:102] [File: odometry_runner.cpp, Line: 102] -- Name: 07, Num Frames=1100 - With  Ground Truth Poses
I20240529 12:15:46.457782 3322870 odometry_runner.cpp:129] Running the Odometry on the sequence named KITTI_07
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append
Caught SIGABRT: usually caused by an abort() or assert()
  message for depth: [2]: /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x63886590c0) [0x7f7800000000]
  message for depth: [3]: /lib/x86_64-linux-gnu/libc.so.6(abort-0x7f1477986729) [0x5]
  message for depth: [4]: /lib/x86_64-linux-gnu/libstdc++.so.6(-0x7f1477cc2000) [(nil)]
  message for depth: [5]: /lib/x86_64-linux-gnu/libstdc++.so.6(+0) [0x7f1477cc2000]
  message for depth: [6]: /lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa300) [0x7f1477d6c300]
  message for depth: [7]: /lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa600) [0x7f1477d6c600]
  message for depth: [8]: /lib/x86_64-linux-gnu/libstdc++.so.6(_ZSt20__throw_length_errorPKc-0x7f1476d531bf) [0x10100e4]
  message for depth: [9]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libSlamCore.so(-0x7f1477f97298) [0xd68]
  message for depth: [10]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libSlamCore.so(_ZN4slam21ReadStreamAsByteArrayERSi-0x4e687c) [0x7f1477b1aff4]
  message for depth: [11]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libSlamCore.so(_ZN4slam21ReadPointCloudFromPLYERSi+0x580f8) [0x7f1478062118]
  message for depth: [12]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libSlamCore.so(_ZN4slam21ReadPointCloudFromPLYERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE-0x7800a5f0) [0x7f1400000010]
  message for depth: [13]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libCT_ICP.so(_ZNK6ct_icp12PLYDirectory9ReadFrameERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x87efe6a0) [0x7f14ffffff20]
  message for depth: [14]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libCT_ICP.so(_ZNK6ct_icp13AFileSequence18GetUnfilteredFrameEm-0x78102e80) [0x7f1400000000]
  message for depth: [15]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libCT_ICP.so(_ZN6ct_icp13AFileSequence19NextUnfilteredFrameEv-0x780fa404) [0x7f140000000c]
  message for depth: [16]: /home/heidrich/wiesn/ct_icp/install/CT_ICP/lib/libCT_ICP.so(_ZN6ct_icp16ADatasetSequence9NextFrameEv-0xa0180) [0x7f1478061d30]
  message for depth: [17]: ./install/CT_ICP/bin/run_odometry(+0x294465427d78) [0x7f1478061d78]
  message for depth: [18]: ./install/CT_ICP/bin/run_odometry(+0x294464ede37d) [0x7f1477b1837d]
  message for depth: [19]: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main-0x7f1477987f90) [(nil)]

boyang9602, 2024-08-28:
Hi,

It works on the provided files. So I guess it is related to the data format. 
1. Convert the data to ply format
2. Make sure the folder structure is as below
```
root@6210a0880c69:/ct_icp/data# tree -L 2
.
|-- 00
|   |-- 00.txt
|   `-- frames
|-- 01
|   |-- 01.txt
|   `-- frames
|-- 02
|   |-- 02.txt
|   `-- frames
|-- 03
|   |-- 03.txt
|   `-- frames
|-- 04
|   |-- 04.txt
|   `-- frames
|-- 05
|   |-- 05.txt
|   `-- frames
|-- 06
|   |-- 06.txt
|   `-- frames
|-- 07
|   |-- 07.txt
|   `-- frames
|-- 08
|   |-- 08.txt
|   `-- frames
|-- 09
|   |-- 09.txt
|   `-- frames
|-- 10
|   |-- 10.txt
|   `-- frames
|-- 11
|   `-- frames
|-- 12
|   `-- frames
|-- 13
|   `-- frames
|-- 14
|   `-- frames
|-- 15
|   `-- frames
|-- 16
|   `-- frames
|-- 17
|   `-- frames
|-- 18
|   `-- frames
|-- 19
|   `-- frames
|-- 20
|   `-- frames
`-- 21
    `-- frames
```
3. In frames, each file is named as frame_<number>.ply

boyang9602, 2024-08-28:
Also, the ply should follow their format as shown by the header below:
```
ply
format binary_little_endian 1.0
element vertex 127097
property float32 x
property float32 y
property float32 z
property float32 timestamp
end_header
```

I am looking for some tools to convert the data. 

If your storage is limited but you do not mind the speed, you can 
1. convert the bin to ply files (it takes several mins)
2. run the experiments
3. delete the ply files

arxidinakbar, 2024-12-23:
@laurenzheidrich  Hi, I had occured the identical error when running, just wonder if you got any solutiuons?
thank you.